MarlinFirmware/MarlinDocumentation #20: Need some good thinking on how to clean up the Z-Probe documentation

Roxy-3DPrintBoard, 2016-04-21:
I know a fair amount of the Auto Bed Leveling.   But, I was trying to bring up RC-5 and I flunked the test.   I got confused by some of the things documented in the Configuration.h file.   Admitably, I should have known better on a some of the mistakes I made.   And any thing I wasn't sure about I could have drilled down on the code and realized I was doing something stupid.

It took ThinkyHead explaining several philosophical points to get me pointed in the right direction.  (The major point he made that helped clear my thinking was "If your electronics has a function labeled on the board, it is defined and called that in the pins.h file for the board."   I was trying to bring up a Delta, and they always home towards Max.  So, I didn't want (for safety reasons) to have a Z-Min defined.

Well...   Just try to get your Z-Probe working without a Z-Min! 

We need better documentation (Both in Configuration.h and the real Documentation) to help people get the Z-Probe configured properly for their printer.   I'm proof of that need.  If I can have trouble getting my Z-Probe working, for sure, other people are going to have trouble!

I don't pretend to know what the answer is.  This whole Z-Probe topic is complicated by a million other factors.   But I do know we need to help the users to guarantee they succeed!

If anybody has some good ideas on how to simplify this issue...  This thread is for that purpose!

sarf2k4, 2016-04-21:
First off, this `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` and `//#define Z_MIN_PROBE_ENDSTOP` quite confusing. 

The `//#define Z_MIN_PROBE_ENDSTOP` in my own opinion should be changed to `enable z min`

This `#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` however is so confusing even the name of the variable itself because the variable name is quite long.

Probably I'm pointing this out at the wrong place

Roxy-3DPrintBoard, 2016-04-21:
> Probably I'm pointing this out at the wrong place

No...   If I had trouble getting the Z-Probe to work, we need some suggestions on how to get this clearly documented.

sarf2k4, 2016-04-22:
Do you have trouble setting up the z-probe before rc4 launched?

Roxy-3DPrintBoard, 2016-04-22:
I'm kind of a special case.   I'm trying to bring up a Delta with a very lame Allen Key probe.  The Allen Key probe needs to be manually deployed.    So right there is a problem.   Even though it looks like an Allen Key, it should not be declared as one because that implies there is a special movement of the nozzle to get it to deploy and retract.

So, that is another item to figure out how to communicate.  We can't be changing things like this in RC-5, but we do need to explain just because it looks like an Allen Key, don't call it that unless it is deployed with special movements.

landodragon141, 2016-10-21:
+Roxy-3D
I edited the documentation text for Z-probe just barely. Does that clear it up/help you?

Roxy-3D, 2016-10-21:
I saw the back and forth between you and ThinkyHead.   I like the changes.

Right now, I'm getting setup to bring up the UBL Bed Leveling system on a gMax 1.5+ printer that has a BLTouch probe on it.   So, I'm going to have another go at setting up a random probe setup.  I'll let you know how it goes!   :)

landodragon141, 2016-11-04:
How did it end up going?

drgrandios, 2018-09-17:
This might be the right place for an improvement request concerning Z-probe configuration. 

The documentation (and most tutorials on the internet) describe that if `Z_MIN_PROBE_ENDSTOP` is used, the `Z_MAX_PIN` would be used as a default pin for the probe.

> On machines that use Z-min for an endstop, the Z-Max pin is recommended next, so this is set as the default alternative on most boards.

This should be either changed or maybe linked to the `pins_[...].h` files because according to this commit
https://github.com/MarlinFirmware/Marlin/commit/9cb596e88c4512da309c580400c4960ff39b0556
most boards use a unique pin for `Z_MIN_PROBE_PIN`.

drgrandios, 2018-09-17:
@thinkyhead - thanks! Where can I find the updated description? Cannot see it at 
http://marlinfw.org/docs/configuration/probes.html
neither at 
https://github.com/MarlinFirmware/MarlinDocumentation/blob/master/_configuration/probes.md

thinkyhead, 2018-09-17:
The verbiage was changed at http://marlinfw.org/docs/configuration/configuration.html

shitcreek, 2019-08-06:
Looks like this has been addressed.

closed 2019-08-06 by shitcreek