MarlinFirmware/MarlinDocumentation #35: https://www.marlinfw.org/ uses invalid cert

gaffo, 2016-08-23:
You are using a *.github.com cert for this site which isn't valid. Browser got an error.

thinkyhead, 2016-08-26:
I suggest we add a redirect to the non-secure domain if https: is used. There is no login or sensitive information associated with the site.

gaffo, 2016-08-26:
That won't work as you'll get the cert warning before the redirect

thinkyhead, 2016-08-28:
.htaccess ftw

gaffo, 2016-08-28:
Is that deployed? it is currently still broken

thinkyhead, 2016-08-28:
Just a suggestion to the maintainer.

jbrazio, 2016-11-26:
The issue is that Github pages only support https using the `.github.io` domain and we're using a custom one. I implemented some time ago a simple js redirect if the user tries to access the site using https because there is no workable workaround so far.

This was the code that was preventing the usage of domains other than our main one.

```js
      /**
       * As Github pages do not support https for custom domains
       * this is the current best workaround for #18.
       */
      var domains  = ['marlinfw.org', 'localhost', '127.0.0.1' ];
      var pathname = location.pathname.replace(/^\/MarlinDocumentation/, '')
      if (domains.indexOf(window.location.hostname) == -1)
          window.top.location.href = 'http://marlinfw.org' + pathname;
```

I've just updated the code do redirect in case of any https request which is sub optimal but no other way to try to ease the UX currently.

closed 2016-11-26 by jbrazio

thinkyhead, 2016-11-26:
I don't think HTTPS will be that important. There's no login. And if the FBI finds browsing marlinfw.org suspicious, we have bigger problems.

gaffo, 2016-11-27:
The things that need to be protected are actually protected by github's cert when doing a clone or https downloads. Seems like an issue with github itself if you can't use a real cert. :( thanks for thinking it through. Sorry if I was a bother.