MarlinFirmware/MarlinDocumentation #63: Consolidation of code for 1284p

Psyc0Flyer, 2017-03-27:
I was wondering what all I can remove from code if I have very specific hardware? I am using a Melzi (Sanguino) board #63, 1X, 1Y, 1Z, 1E, BLTouch and a RepRap Discount Full Graphics Display? I want to have all of the functions of the most current Marlin but a Jenny Craig version. Any and all help would be greatly appreciated!

TIA!

thinkyhead, 2017-03-27:
There is no need to strip out code. Marlin only compiles in the code it needs for the features you have enabled. To get a smaller build, turn off the features you don't need.

Psyc0Flyer, 2017-03-27:
That is what I am asking. I have only selected what I need (I think) and it tells me it is over by 6002 bytes :/

fiveangle, 2017-05-31:
setting NO_WORKSPACE_OFFSETS in advance config is usually an easy way to get a few k back

`#define NO_WORKSPACE_OFFSETS`

thinkyhead, 2017-06-15:
See if the new `bugfix-1.1.x` is slimmer. We've done a bunch of optimization to try and reduce SRAM usage. 6002 bytes over sounds very strange. Even with a full configuration, Marlin should only use around 5K of SRAM.

fiveangle, 2017-07-15:
It's a 128k board so I'm sure he's talking about PROGMEM. Surprised it's only 6k over with all that stuff enabled. Disabling workspace offsets, cold extrusion prevention, disable 503, and maybe a few other small things should get you there.

closed 2017-11-24 by thinkyhead