MarlinFirmware/MarlinDocumentation #80: Saving UBL Mesh and reload after flashing new firmware

fr33dimension, 2017-05-30:
Is there a way to save the mesh data before flashing a new firmware and then reload it?
It's very time consuming to redo the mesh anytime you flash.

Roxy-3D, 2017-05-30:
Yes...   You can save a GCode file to your computer by doing a G29 S-1   (saving to slot -1).   Cut & Paste those GCode lines to a file.   You can print that file to your printer to reload the mesh if incompatibilities pop up across firmware versions.

fr33dimension, 2017-05-31:
That is very nice. I am able to save the g code starting with G29 I 999, then M421 I 0 J 0 Z -0.138438 ; X 0.00, Y 0.00 and so on....... When printing this it loads it alright, but UBL is not working. In other words, I can load or restore the mesh data, save it to slot 1 or whatever, activate the system and nothing....what am I doing wrong?

fr33dimension, 2017-05-31:
Got it! Great.

Nonametruck, 2017-06-03:
How often should UBL be performed once we store a good mesh??

Roxy-3D, 2017-06-03:
The mesh can be stored in the EEPROM and it can be stored on your computer.   The mesh saved on your computer will become less and less important.   Mostly, it is there so we can shift the way things are stored in the EEPROM and have an easy way to recover the user's data.    But some people have multiple printers and occasionally want to move a piece of glass from one printer to another.   In that case, being able to save the mesh on a computer is very helpful.

> How often should UBL be performed once we store a good mesh??

Other than for testing reasons, I have always operated with UBL active at start up.   It loads the mesh and it is working even though I don't even have a G29 in my Slicer's start up code.   You do that by getting a mesh one way or the other.   Save the mesh to a particular slot number with G29 S1.  Set your fade height (probably G29 F 10.0).    Activating the system with a G29 A.   And then saving your EEPROM settings with a M500.

At that point...  You don't need to do any more G29's until something changes.

closed 2017-08-28 by thinkyhead