MarlinFirmware/MarlinDocumentation #85: Docco or Code issue with Z_MIN_PROBE_ENDSTOP

apelly, 2017-06-23:
While working through ```Configuration.h``` setting up UBL without a probe I encountered some behaviour that took me ages to track down.
```
/**
 * Z_MIN_PROBE_ENDSTOP
 *
 * Enable this option for a probe connected to any pin except Z-Min.
 * (By default Marlin assumes the Z-Max endstop pin.)
 * To use a custom Z Probe pin, set Z_MIN_PROBE_PIN below.
```
There seem to be a couple of issues here.

1. I don't think it does default to ```Z_MAX_PIN```
   The reason I don't think it does is that I get a compile time error:
   ```
   In file included from /tmp/arduino_build_434346/sketch/Marlin_main.cpp:308:0:
   /tmp/arduino_build_434346/sketch/endstop_interrupts.h: In function 'void    setup_endstop_interrupts()':
   endstop_interrupts.h:198: error: static assertion failed: Z_MIN_PROBE_PIN is not interrupt-capable
          static_assert(digitalPinToPCICR(Z_MIN_PROBE_PIN) != NULL, "Z_MIN_PROBE_PIN is not interrupt-capable");
          ^
   ```
   When I add
   ```
   #define Z_MIN_PROBE_PIN Z_MAX_PIN
   ```
   This compile issue goes away.
2. In the section Endstop Settings the docco _does_ say
   ```
   // Specify here all the endstop connectors that are connected to any endstop or probe.
   ```
   but that is in a different section of the config file. It took me a hell of a lot of grepping to realise that is what was missing from my config, causing more errors. I think it'd be nice to mention this in the ```Z_MIN_ENDSTOP``` comment, and/or upgrade the sanity check, which fails with 
   ```
   #error "Z_MIN_PROBE_ENDSTOP requires the Z_MIN_PROBE_PIN to be defined."
   ```
   because ```pins.h``` says
   ```
   #if DISABLED(USE_ZMAX_PLUG)
     #undef Z_MAX_PIN
     #define Z_MAX_PIN          -1
   #endif
   ```

Bob-the-Kuhn, 2017-06-23:
`#define Z_MIN_PROBE_PIN Z_MAX_PIN` was removed from Configuration.h because it doesn't work.  It results in the Z_MIN_PROBE_PIN not having a value.  That results in the "interrupt-capable" test not being applied to the Z_MIN_PROBE_PIN.  Strange stuff happens at compile time.
       
`#define Z_MIN_PROBE_PIN 18` in either Configuration.h or in the pins_YOUR_BOARD.h file works as expected.

As best I can tell, you are correct that there is NOT a default for the Z_MIN_PROBE_PIN.   I can't find one in the code and you get a sanity check error when Z_MIN_PROBE_ENDSTOP is enabled but Z_MIN_PROBE_PIN isn't specified.

I'll start a PR to correct the defaults to Z_MAX statement in the Configuration.h files.

Bob-the-Kuhn, 2017-06-23:
Definitely past my bed time.  

The offending statement is in the probes document on the Wiki.

apelly, 2017-06-23:
It's late for me too, but I think you are describing what happens if ```Z_MIN_ENDSTOP``` is not defined.

I spent a whole afternoon looking at this, and there is a default. In some situations. The trouble is the other _X_ (re)definitions in some unknown hierarchy screwing with what I _declared_.

Bob-the-Kuhn, 2017-06-23:
> screwing with what I declared

If you can provide examples of this I can investigate & see if there is a reasonable explanation.

Bob-the-Kuhn, 2017-06-23:
PR #86 has been created to delete the Z-MAX default line.

apelly, 2017-06-23:
As I said: Despite the semi-policy of limiting the words in the source, I think it would be kind to mention ```Z_MIN_ENDSTOP``` in that comment.

I _did_ see evidence that something tries to use the z-max pin, but it was obviously from some time ago, because there is a lot of other conflicting code now.

As for redefining things; there's a bunch of stuff related to the bl-touch somewhere obtuse, a bunch in conditional_lcd.h, pins_xxx.h does it too.

I'll raise a separate issue if you think it's worth it.

Bob-the-Kuhn, 2017-06-23:
I know all about the BLTouch defines.  We did it that way to make life simpler for the users which also makes supporting Marlin a lot easier.  Unless there's an error or an unusual situation we'll leave them alone.

apelly, 2017-06-23:
Sweet. I was just pointing out some things that caused me pain or looked odd to me.

closed 2017-06-23 by apelly