MarlinFirmware/MarlinDocumentation #126: Is M501 necessary in the M502, M500, M501 initialization sequence?

filipgoc, 2018-01-24:
Hi, the [UBL EPPROM initialization](http://marlinfw.org/docs/gcode/G029-ubl.html) sequence is 

```
M502          ; Reset settings to configuration defaults...
M500          ; ...and Save to EEPROM. Use this on a new install.
M501          ; Read back in the saved EEPROM.
```

Isn't M501 redundant? 
Is it just to let the user verify what's in EEPROM now visually, but not actually needed?

Cheers.

filipgoc, 2018-01-30:
I did a bit more searching, and it does seem that the documentation is wrong here. 

Quotink Thinkyhead in [this post](https://github.com/MarlinFirmware/Marlin/issues/3233)
> M500 followed by M501 should have no effect. You're only reading back the settings just written. 

> M500 writes the current settings (in RAM) to EEPROM, so M501 immediately after M500 only reads back what you just wrote.

So... maybe I'll submit a PR, but I'll let it hang here for a bit longer to get some rebuttals.

thinkyhead, 2018-02-06:
This was a requirement of UBL because it didn't know how many mesh slots were available in the EEPROM until the `settings.load()` function supplied the count. I did a phat update of the EEPROM code, adding a proper data structure, and now the `M501` is not required for UBL to know how many slots it has!

closed 2018-02-23 by thinkyhead