MarlinFirmware/MarlinDocumentation #141: Add available information as appendix to UBL documenation

Hugoz12, 2018-03-11:
As I removed my capacitive sensor to go with manual probing I needed more information about the parameters of the P2 commend (to get a better understanding).
Finding it in ubl_G29.cpp was very useful. Why not add these information as an appendix to the UBL documentation?
Or give a hint where to find it.

This is the info I'm referring to:
```*   G29: Unified Bed Leveling by Roxy
   *
   *   Parameters understood by this leveling system:
   *
   *   A     Activate   Activate the Unified Bed Leveling system.
   *
   *   B #   Business   Use the 'Business Card' mode of the Manual Probe subsystem with P2.
   *                    Note: A non-compressible Spark Gap feeler gauge is recommended over a business card.
   *                    In this mode of G29 P2, a business or index card is used as a shim that the nozzle can
   *                    grab onto as it is lowered. In principle, the nozzle-bed distance is the same when the
   *                    same resistance is felt in the shim. You can omit the numerical value on first invocation
   *                    of G29 P2 B to measure shim thickness. Subsequent use of 'B' will apply the previously-
   *                    measured thickness by default.
   *
   *   C     Continue   G29 P1 C continues the generation of a partially-constructed Mesh without invalidating
   *                    previous measurements.
   *
   *   C     Constant   G29 P2 C specifies a Constant and tells the Manual Probe subsystem to use the current
   *                    location in its search for the closest unmeasured Mesh Point.
   *
   *                    G29 P3 C specifies the Constant for the fill. Otherwise, uses a "reasonable" value.
   *
   *   C     Current    G29 Z C uses the Current location (instead of bed center or nearest edge).
   *
   *   D     Disable    Disable the Unified Bed Leveling system.
   *
   *   E     Stow_probe Stow the probe after each sampled point.
   *
   *   F #   Fade       Fade the amount of Mesh Based Compensation over a specified height. At the
   *                    specified height, no correction is applied and natural printer kenimatics take over. If no
   *                    number is specified for the command, 10mm is assumed to be reasonable.
   *
   *   H #   Height     With P2, 'H' specifies the Height to raise the nozzle after each manual probe of the bed.
   *                    If omitted, the nozzle will raise by Z_CLEARANCE_BETWEEN_PROBES.
   *
   *   H #   Offset     With P4, 'H' specifies the Offset above the mesh height to place the nozzle.
   *                    If omitted, Z_CLEARANCE_BETWEEN_PROBES will be used.
   *
   *   I #   Invalidate Invalidate the specified number of Mesh Points near the given 'X' 'Y'. If X or Y are omitted,
   *                    the nozzle location is used. If no 'I' value is given, only the point nearest to the location
   *                    is invalidated. Use 'T' to produce a map afterward. This command is useful to invalidate a
   *                    portion of the Mesh so it can be adjusted using other UBL tools. When attempting to invalidate
   *                    an isolated bad mesh point, the 'T' option shows the nozzle position in the Mesh with (#). You
   *                    can move the nozzle around and use this feature to select the center of the area (or cell) to
   *                    invalidate.
   *
   *   J #   Grid       Perform a Grid Based Leveling of the current Mesh using a grid with n points on a side.
   *                    Not specifying a grid size will invoke the 3-Point leveling function.
   *
   *   K #   Kompare    Kompare current Mesh with stored Mesh # replacing current Mesh with the result. This
   *                    command literally performs a diff between two Meshes.
   *
   *   L     Load       Load Mesh from the previously activated location in the EEPROM.
   *
   *   L #   Load       Load Mesh from the specified location in the EEPROM. Set this location as activated
   *                    for subsequent Load and Store operations.
   *
   *   The P or Phase commands are used for the bulk of the work to setup a Mesh. In general, your Mesh will
   *   start off being initialized with a G29 P0 or a G29 P1. Further refinement of the Mesh happens with
   *   each additional Phase that processes it.
   *
   *   P0    Phase 0    Zero Mesh Data and turn off the Mesh Compensation System. This reverts the
   *                    3D Printer to the same state it was in before the Unified Bed Leveling Compensation
   *                    was turned on. Setting the entire Mesh to Zero is a special case that allows
   *                    a subsequent G or T leveling operation for backward compatibility.
[... 135 lines cut ...]
   *   kinds of problems. Enabling EEPROM Storage is highly recommended. With EEPROM Storage
   *   of the mesh, you are limited to 3-Point and Grid Leveling. (G29 P0 T and G29 P0 G
   *   respectively.)
   *
   *   When you do a G28 and then a G29 P1 to automatically build your first mesh, you are going to notice
   *   the Unified Bed Leveling probes points further and further away from the starting location. (The
   *   starting location defaults to the center of the bed.)   The original Grid and Mesh leveling used
   *   a Zig Zag pattern. The new pattern is better, especially for people with Delta printers. This
   *   allows you to get the center area of the Mesh populated (and edited) quicker. This allows you to
   *   perform a small print and check out your settings quicker. You do not need to populate the
   *   entire mesh to use it. (You don't want to spend a lot of time generating a mesh only to realize
   *   you don't have the resolution or zprobe_zoffset set correctly. The Mesh generation
   *   gathers points closest to where the nozzle is located unless you specify an (X,Y) coordinate pair.
   *
   *   The Unified Bed Leveling uses a lot of EEPROM storage to hold its data. And it takes some effort
   *   to get this Mesh data correct for a user's printer. We do not want this data destroyed as
   *   new versions of Marlin add or subtract to the items stored in EEPROM. So, for the benefit of
   *   the users, we store the Mesh data at the end of the EEPROM and do not keep it contiguous with the
   *   other data stored in the EEPROM. (For sure the developers are going to complain about this, but
   *   this is going to be helpful to the users!)
   *
   *   The foundation of this Bed Leveling System is built on Epatel's Mesh Bed Leveling code. A big
   *   'Thanks!' to him and the creators of 3-Point and Grid Based leveling. Combining their contributions
   *   we now have the functionality and features of all three systems combined.
   */

thinkyhead, 2018-03-11:
Agreed. The documentation should be more filled out on the website. We should add this synopsis (with nice formatting) to the http://marlinfw.org/docs/features/unified_bed_leveling.html page.

Roxy-3D, 2018-03-12:
At least nobody is complaining the code isn't documented!  :)  

Help from the user base to get this into the user documentation would be much appreciated.

thinkyhead, 2018-03-12:
Sounds like it's going to be me who does this one. Will take care of it now.

closed 2018-03-12 by thinkyhead

thinkyhead, 2018-03-12:
> You MUST do M502, M500 to initialize the storage.

This might no longer be required, or is it? If the base EEPROM test fails (checksum, version) is UBL load skipped? Does (or could) UBL look for a valid key or checksum when loading its data, so it won't load uninitialized the random bytes in EEPROM (and maybe write the current mesh to initialize it)?

> With EEPROM Storage of the mesh, you are limited to 3-Point and Grid Leveling.

Should this be "Without…"? It sounds like enabling EEPROM imposes extra limits.

> (For sure the developers are going to complain about this, but this is going to be helpful to the users!)

Since no one has complained (that I am aware of), can we change this to "_Happily, no developers have complained about UBL borrowing the upper half of EEPROM for its nefarious purposes_"?

thinkyhead, 2018-03-12:
I've applied the longer descriptions above directly to the [`G29` UBL page](http://marlinfw.org/docs/gcode/G029-ubl.html) in the parameter descriptions. If desired, I can tweak the CSS to space things out or improve readability of the longer bullet items.

Roxy-3D, 2018-03-12:
@thinkyhead   Is it possible to make changes be reflected immediately?  One of the problems is when changes are made, they don't show up until some other special action is done.  And I don't know how to make that happen.  (And I don't have the tools installed to make that happen.)

>> You MUST do M502, M500 to initialize the storage.

> This might no longer be required, or is it? If the base EEPROM test fails (checksum, version) is UBL loading skipped? Does (or could) UBL look for a valid key or checksum when loading its data, so it won't load uninitialized the random bytes from EEPROM (and maybe write the current mesh to initialize it)?

We should probably put CRC's on each the data for each mesh.   But independent of this...   I think there is a problem where the EEPROM has to be initialized with the new version's data or something causes the nozzle to crash into the bed.   I think that is still present.  I have no way of testing that right now.

>> With EEPROM Storage of the mesh, you are limited to 3-Point and Grid Leveling.

> Should this be "Without…"? It sounds like enabling EEPROM imposes extra limits.

Yes.   This is backwards as documented.

>> (For sure the developers are going to complain about this, but this is going to be helpful to the users!)

> Since no one has complained (that I am aware of), can we change this to "Happily, no developers have complained about UBL borrowing the upper half of EEPROM for its nefarious purposes"?

Agreed!

thinkyhead, 2018-03-14:
> I don't have the tools installed to make that happen.

First thing is to make a fork of this repo under your account. Then install Jekyll on your computer. Instructions for installing Jekyll (the first major step) are posted in the README on the front page: https://github.com/MarlinFirmware/MarlinDocumentation. An installation of Ruby and Ruby Gems is required to use Jekyll.

Documentation edits are made directly to the `master` branch — in a hybrid of YML and Markdown. When previewing the documentation locally, the HTML is built in-place and stashed in a folder that's ignored by `.gitignore`, and then Jekyll runs a little webserver at 127.0.0.1:4000.
```
bundle exec jekyll serve --watch --incremental
```
When edits are made to files and saved, Jekyll updates the edited pages right away in the preview site. Typically after saving an edited page it takes Jekyll 2-4 seconds to produce the new HTML. The whole site functions, including the scripts.

I use the `mfdoc` and `mfpub` scripts in the Marlin `buildroot/share/git` folder to preview and publish the documentation. They require a GNU BASH environment to run. Even without a GNU shell, they are still instructive to read, since they show the steps required to publish the built site. The `mfpub` script is careful to pull the latest changes from the `master` branch on Github and to push local changes to both `origin` and `upstream` so everything is in sync. The last step in publishing the HTML is to push the modified `gh-pages` branch to `upstream`.

> Yes. This is backwards as documented.

I'll fix that soon!

Roxy-3D, 2018-03-15:
I'll be home in a few days...  But it will take a few more days before I can mess with things.    I'll see if I can get Jekyll to do the right thing.   But I'm on a Windoze machine so it may be a little bit harder to get things working.   Hopefully I won't have to send up a signal flare to you and Joao to fix things...

Can the documentation branch be made to automatically run those tools when a Pull Request is merged?

thinkyhead, 2018-03-15:
> Can the documentation branch be made to automatically run those tools when a Pull Request is merged?

Probably not. Publishing to the branch requires credentials. But it's not vital. I always push the changes right after I clean up and merge any PR.

The bulk of work needed right now is to complete and update the configuration page.