MarlinFirmware/MarlinDocumentation #142: how do I auto power e fans?

ruggb, 2018-03-12:
I have the auto power enabled but I don't use a ATX PSU.
I have my e fan connected to D11 but I can't figure out how to define it and make it auto work.
There is no definition for it in pin_Ramps.h that I see.
I can manually turn it on/off with M42.

thinkyhead, 2018-03-12:
This Issue Queue is for reports related to Marlin documentation-related issues and contributions, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following resources:
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)

closed 2018-03-12 by thinkyhead

ruggb, 2018-03-12:
I guess I would not have asked the question if it WERE DOCUMENTED?

thinkyhead, 2018-03-14:
I'm going to let others answer your questions from now on.