MarlinFirmware/MarlinDocumentation #158: Where to put documentation on how to use G26 to physically level the bed?

ManuelMcLure, 2018-05-05:
I want to write a documentation section on how to use G26 to help physically level the bed by moving to each of the four corners and running a partial G26 (producing a 2x2 subgrid) at each corner, for example:

```G26 D X0 Y0 R4 B65 H205 L0.2 O3
G26 D X220 Y0 R4 B65 H205 L0.2 O3
G26 D X220 Y220 R4 B65 H205 L0.2 O3
G26 D X0 Y220R4 B65 H205 L0.2 O3
```

My first thought was to add this to the UBL doc, but this really applies to any of the leveling methods that can enable `G26_MESH_VALIDATION` so I'm not sure where the best place to put it is. Any suggestions?

On a different note, I was thinking of possibly adding an option to G26 that would automate generating this pattern. I'm trying to figure out which letter would be the best for this option - it looks like E I M N T V W Z are still available. Do you think this would be a good addition to G26, and which letter would be the best? Unfortunately P for physical and C for corner are already used. Maybe T for tramming would work.

Roxy-3D, 2018-05-05:
You can remove the U option (and its code) and use that for 'Unlevel Beds'.

ManuelMcLure, 2018-05-06:
I'm working on using the arc command (calling `plan_arc()`) and am seeing something that concerns me. It looks like `plan_arc()` is calling `planner.buffer_line_kinematic()` directly and unless I'm missing something `planner.buffer_line_kinematic()` does not apply any leveling unless `PLANNER_LEVELING` is set. `PLANNER_LEVELING` is defined as
```
#define UBL_SEGMENTED  (ENABLED(AUTO_BED_LEVELING_UBL) && (ENABLED(DELTA)))
#define ABL_PLANAR     (ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_3POINT))
#define ABL_GRID       (ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR))
#define OLDSCHOOL_ABL  (ABL_PLANAR || ABL_GRID)
...
#define PLANNER_LEVELING      (OLDSCHOOL_ABL || ENABLED(MESH_BED_LEVELING) || UBL_SEGMENTED || ENABLED(SKEW_CORRECTION))
```
so as far as I can tell `PLANNER_LEVELING` is **not** set if using UBL on a cartesian printer and therefore it looks like leveling is not applied to arc moves. If I'm missing something please let me know.

Roxy-3D, 2018-05-06:
Yikes!    I think you are correct.    It might be we should have a #if ENABLED() for the UBL non-segmented case to send it down the correct path.

THANK YOU for finding this!

ManuelMcLure, 2018-05-06:
It looks like bezier curves have the same issue.

Roxy-3D, 2018-05-06:
Yes, please open an issue.   (But in the Marlin/Firmware area...   Otherwise it might not get the attention that it needs.)

thinkyhead, 2018-05-08:
> It looks like bezier curves have the same issue.

Modify the end of `plan_arc`…
```diff
  // Ensure last segment arrives at target location.
  #if ENABLED(SCARA_FEEDRATE_SCALING)
    inverse_kinematics(cart);
    ADJUST_DELTA(cart);
    const float diff2 = HYPOT2(delta[A_AXIS] - oldA, delta[B_AXIS] - oldB);
    if (diff2)
      planner.buffer_segment(delta[A_AXIS], delta[B_AXIS], cart[Z_AXIS], cart[E_AXIS], SQRT(diff2) * inverse_secs, active_extruder);
+ #elif ENABLED(AUTO_BED_LEVELING_UBL) && !PLANNER_LEVELING
+   float pos[XYZ] = { cart[X_AXIS], cart[Y_AXIS], cart[Z_AXIS] };
+
+   #if ENABLED(SKEW_CORRECTION)
+     skew(pos[X_AXIS], pos[Y_AXIS], pos[Z_AXIS]);
+   #endif
+
+   if (leveling_active) {
+     #if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
+       const float fade_scaling_factor = planner.fade_scaling_factor_for_z(rtarget[Z_AXIS]);
+     #else
+       constexpr float fade_scaling_factor = 1.0;
+     #endif
+     const float cz = pos[Z_AXIS] + (fade_scaling_factor ? fade_scaling_factor * ubl.get_z_correction(pos[X_AXIS], pos[Y_AXIS]) : 0.0);
+   }
+   planner.buffer_segment(pos[A_AXIS], pos[B_AXIS], cz, cart[E_AXIS], fr_mm_s, active_extruder);
  #else
    planner.buffer_line_kinematic(cart, fr_mm_s, active_extruder);
  #endif
```

closed 2019-08-07 by shitcreek