MarlinFirmware/MarlinDocumentation #159: Origin vs Home position vs ...?

filipgoc, 2018-05-10:
I think the way Marlin and more importantly the docs are using '**origin**' is confusing and needs to be clarified. 

## background: 

- In the documentation, 'origin' is used for G28 position, as in `G28 ; Go to origin on all axes`. 
- In Marlin language files, the MSG_SET_ORIGIN is a somewhat abandoned thing that called up `G92 X0 Y0 Z0`, but menu-wise replaced by MSG_SET_HOME_OFFSETS with `M428`.  (And we do not usually associate G92 with 'origin' so... that was probably some early inconsistency.) 

## reasoning

**I think that G28 aftermath should not be called 'origin'.** Even though this is fundamentally the origin for the absolute space, we do not expose this origin to the users! Instead, we allow home offsets (and other things) to shift the functional [0,0] around to tweak the actual starting position. (On a cartesian printer, this would correspond to the corner of the build area.)

From the user's perspective, it makes much more sense to refer to the [0,0] as affected by home offsets are the origin. This is the 'origin' of the build area and the start of the print. And this is _also_ what we actually display on the LCD as [0,0]! Surely, it would make sense to refer to [0,0] the origin - or at least it's the tradition. 

## summary proposal: 

- **Home position**
   - Use Home position for whatever happens when G28 is run
- **Origin**
   - reserve Origin for the [0,0] as affected by home offsets and/or g92
- in terms of documentation, this means to get rid of the word 'origin' from the G28 help page. 
- ideally, it would also mean to rename the MSG_SET_ORIGIN, but that does seem to be a hassle that is maybe unnecessary.

filipgoc, 2018-05-10:
PS: I'm curious how do you guys explain this to newbies; what language do you use?

If you use 'origin' for the G28 position as the docs do, what do you use to refer to the actual print origin that matches the origin in slicer?

I'm happy to adopt whatever language... but I don't want everyone to be using the same words to mean something else.

thinkyhead, 2018-05-11:
The term **"origin"** can be entirely dropped. If used to refer to the homing position, then it should be replaced by **"home"**. It the word "origin" is used in other ways, replace it with whatever makes the most sense.

When a slicer refers to "origin" it usually means X0 Y0 Z0. On a delta this "origin" is the center of the bed. On a Cartesian, it is one of the corners, usually front-left, though there's nothing preventing a Cartesian from making the center of the bed X0 Y0.

Since there are no moves in Marlin that "go to origin" or that "set the origin," and there's no serial output or menus that refer to "origin," we can drop the term entirely from the documentation.

filipgoc, 2018-05-11:
@thinkyhead alrite. I'll make it an opportunity to finally set up local docs environment for myself. Probably will take a few days, but it's not like this is an emergency...

closed 2018-05-11 by filipgoc