MarlinFirmware/MarlinDocumentation #162: M118 is documented on RepRapWiki but not here

b-morgan, 2018-05-17:
Documentation for [M118](http://reprap.org/wiki/G-code#M118:_Echo_message_on_host) is missing from here.

Roxy-3D, 2018-05-17:
It's some what documented at the top of Marlin_main.cpp:
```
 * M117 - Display a message on the controller screen. (Requires an LCD)
 * M118 - Display a message in the host console.
 * M119 - Report endstops status
```

b-morgan, 2018-05-18:
Then what is the page [http://marlinfw.org/meta/gcode/](url) for? Most people use that page to determine what Gcodes are supported by Marlin. Most people don't read the sources. I thought this was an issue against the documentation, not the source code. Am I wrong?

Roxy-3D, 2018-05-18:
Yes, it should be documented better.   I was just saying there is a tiny amount of documentation 'here'.

filipgoc, 2018-05-19:
@b-morgan This is an open source project. When you discover something, it's often an omission rather than a design flaw...

b-morgan, 2018-05-19:
@filipgoc I understand that and if the page in question wasn't so complicated, I'd have attempted to fix it myself.

thinkyhead, 2018-05-24:
Thanks for the heads-up @b-morgan! I've now added a page for `M118` to the G-code docs.

closed 2018-05-24 by thinkyhead