MarlinFirmware/MarlinDocumentation #176: Switch to PlatformIO IDE for VSCode

ivankravets, 2018-09-10:
Firstly, I want to say a big THANK YOU for using @PlatformIO! I personally impressed with your multi-platform configuration https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/platformio.ini

We have a huge plan for [PlatformIO Core 4.0](https://github.com/platformio/platformio-core/milestone/57). I hope that https://github.com/platformio/platformio-core/issues/1643 will singnificnatly simplify `platformio.ini`.

-----

We have recently announced the [massive release of PlatformIO IDE for VSCode](https://community.platformio.org/t/massive-release-of-platformio-ide-for-vscode/4929). We [received great feedback](https://marketplace.visualstudio.com/items?itemName=platformio.platformio-ide#review-details)!

Also, we droped Atom from our main page and now forward all people to PlatformIO IDE for VSCode. I would be thankful if someone could help with updating this documentation: http://marlinfw.org/docs/basics/install_platformio.html

I'm not expert in Marlin Firware but have just tried to use it with our latest version for VSCode and it is MUCH MUCH simpler than for Atom. What I did?

1. Installed [PlatformIO IDE for VSCode](https://platformio.org/install/ide?install=vscode)
2. Switch to new PlatformIO View in Activity Bar (left side bar), then Quick Access > Clone Git Project ...
[image]
3. VSCode clonned whole repo and I received access to all branches and tags! Great! I tried different, `bugfix-2.0x` works great!
[image]
4. Then great addition to our extension for VSCode, that it has Project Task Explorer. Now, no need to edit `platformio.ini` and setup `env_default`, developers can work with multi-environments directly from IDE:
[image]

----------

ModMike, 2018-09-10:
You are right, that is much easier! I've been struggling with Atom.  This looks like a 1 stop shop.

ivankravets, 2018-09-10:
And what is interesting, no need to redownload project again. Just press PULL and new changes are here. It's also good to track changes, etc.

ModMike, 2018-09-10:
Ok so work flow is:

1) Fork project
2) Clone project
3) Select branch

How do I check out the branch?

ModMike, 2018-09-10:
Just realized, it's checked out the moment I select the branch, correct?

ivankravets, 2018-09-10:
@ModMike , yes, VSCode clones whole repo and change branch to "default" which is set on Github. Nevertheless, you can switch to another branch, see left bottom corner.

thinkyhead, 2018-09-11:
Thanks for the early notice, @ivankravets. The multi-platform and Auto Build options for Marlin are the result of great work by @Bob-the-Kuhn and others, and make the build process a lot easier.

I'm happy to see more emphasis on Visual Studio Code, as I'm a big fan of the application and have been evangelizing it for its technical superiority over Atom (which _still_ cannot do multi-line search on multiple files 😕 ).

Of course, I expect to continue using fast-and-lean Sublime Text 3 for my everyday editing. But it sounds like VSCode is going to be a very good alternative to GitHub Desktop going forward.

---
As for your last post, I presume I can drop my existing working copy folder onto VSCode and it will work as before, and won't switch back to the default (1.1.x) branch.

Anyway, I'll give the latest VSCode + PlatformIO a test tomorrow, and I'll aim to get our build documentation updated as soon as possible to put VSCode more front-and-center, while still keeping the Atom build instructions available for holdouts.

ivankravets, 2018-09-12:
> I presume I can drop my existing working copy folder onto VSCode and it will work as before, and won't switch back to the default (1.1.x) branch.

Sure, this option is also allowed. I showed above really complicated use case when a user decides to work with upstream version and be able to pull new changes without redownloading ZIP.

ivankravets, 2019-07-04:
Friends, could we recommend Marlin users to use PlatformIO + VSCode? It has much better support for C/C++ development and includes free debugging. We plan to make a warning for Atom users and ask them to take a look at VSCode for better productivity and performance.

I see that people come to PlatformIO from this page http://marlinfw.org/docs/basics/install_platformio.html

We would be thankful if we can update this page and drop Atom. If you need something from our side, please ping me here.

Thanks!

thinkyhead, 2019-07-05:
I've been telling everybody I can for about a year now, but yes we do need to get the documentation back into shape. I did a nice job of stitching together the Atom and VSCode instructions so that everything was covered, but somewhere along the way that verbiage got lost in the avalanche.

I believe @shitcreek is on the case and getting some nice screenshots together as we speak….

sl1pkn07, 2019-07-09:
this change affect to use platformio from CLI?

ivankravets, 2019-07-09:
The CLI has not been changed since the first 1.0.0 release. We try to keep it compatible between major upgrades. Even upcoming PlatformIO Core 4 is fully compatible with PlatofrmIO Core 3.0 projects.

shitcreek, 2019-07-24:
Updated install PlatformIO instructions as per your suggestion. I took the liberty and credited you as a contributor in the article. Now awaiting rebuild of site.

#4798a60fdcd0a149f6ce4f1cc6c04abfd2a66c09
#d59bffa993899a9d862e340bdf116c8f47517f91

closed 2019-07-24 by shitcreek

ivankravets, 2019-07-25:
Thanks you so much!!! 🤗