MarlinFirmware/MarlinDocumentation #203: FYSETC v1.0 driver jumper configuration image confusion

kc8rwr, 2019-01-09:
The image http://marlinfw.org/assets/images/docs/hardware/tmc_drivers/FYSETC_tmc2130._SPI.jpg found on the page http://marlinfw.org/docs/hardware/tmc_drivers.html is confusing or perhaps wrong.  

This is an image of the FYSETEC v1.0 driver outlining it's three jumpers and explaining how to set them for SPI mode. The problem is that one larger jumper is clearly closed but the label over it says open.  The other two smaller jumpers are open but the label over them says closed! Which is it supposed to be?

Perhaps the image indicates how the driver comes from the factory and the labels injdicate that all three jumpers need to be changed to the opposite of their factory settings to get SPI mode.  This isn't very clear in the text. The text does say that the big jumper to the left of the chip needs to be opened but for the other two it just says that they need to be configured "as shown".

Assuming that what is meant is "open the big jumper which comes closed and close the two little jumpers that come open from the factory" then it would be much more clear if either the image showed the "after" picture instead of the "before" or if the text stated everything explicitly. "Open the big jumper and close the two little ones". As it is I think someone installing one of these drivers for the first time might be a little nervous from this ambiguity and be forced to go look up the datasheet before proceeding.

thinkyhead, 2019-03-13:
Thanks for the report. I will try to find the person who wrote that article and supplied the image to see if they can improve both.

shitcreek, 2019-08-06:
Updated the image to address any confusion.

closed 2019-08-06 by shitcreek