MarlinFirmware/MarlinDocumentation #223: Marlin Bootloader

foxabilo, 2019-06-03:
I wanted to see if my experience as a creator of new controller was consistent with how the Marlin team think it should be, I designed a new controller using the LPC1768 and adopted the SKR v1.3 pinouts and because my MCU chips are from the manufactures they needed to be flashed with a suitable bootloader, for that I used the smoothieware precompiled binary and it worked to successfully change the FIRMWARE.BIN to FIRMWARE.CUR on the SD card with a Marlin 2.0 bugfix  binary from the platform.io environment. 

One thing that does not seem to be documented is the smoothieware bootloader part as that is assumed to come preloaded, is there an official "Marlin" bootloader or is the Smoothieware the way to go and should that be in the Documentation for installation? even as a side note.

thinkyhead, 2019-06-14:
That would be a good addition!

shitcreek, 2019-07-30:
Added instructions for generating firmware.bin
e7e8ac7878e65c98076619404ae3b05a7855795a

closed 2019-07-30 by shitcreek