MarlinFirmware/MarlinDocumentation #224: Auto-Build instructions?

bwaslo, 2019-06-08:
I'm trying to set up to put Marlin 2.0 onto an SKR-V1.3 board, using PlatformIO (instructions at http://marlinfw.org/docs/basics/install_platformio.html) in Atom.
But running into problems at the "Build Marlin" section:
"**Open the top level Marlin directory in PlatformIO.
Click on the “Auto Build” menu at the right end of the main menu bar to bring up the dialog.**"
Where is this "Auto Build" menu (or its Python script file for that matter)?  I followed all instructions, but there's no such menu showing in my Atom/PlatformIO installation.  What am I missing?
Can't find any reference to "Auto-Build" by  searching the PlatformIO docs either.  Been at this a while now..

Bob-the-Kuhn, 2019-06-08:
99.9% of the time it's one of these:
- Marlin directory is not open.  You should see a layout similar to 4D under **Open Marlin in PlatformIO IDE**.
- **process-pallette** was not installed (step 3 under **Install Atom, process-palette and PlatformIO**)

I'm sporadically working on a video version of the instructions.  It has a simpler way of opening the Marlin directory.  You would be doing me a favor if you tried the [rough video](https://www.dropbox.com/s/qn1cvxf9xhdt3x8/Marlin%20Auto%20Build%20video%20%28rev%201%29.mp4?dl=0) and gave me feedback.

bwaslo, 2019-06-08:
Thanks for the quick reply,  And I just found it.   I had the directory named "Marlin" open, apparently should've been its parent directory "Marlin-bugfix-2.0.x", as generated when I extracted the zip file from this site.  Now the "Auto-build" menu shows up.

Bob-the-Kuhn, 2019-06-08:
Please close this issue after you've been able to build & download.

bwaslo, 2019-06-08:
I don't have the new board yet, so no loading it to the target yet, I'm afraid.  Just preparing for when it comes in about a month from now..
I looked at your video, and tried the "Auto-Build" menu anyway per the video, but different from that, was presented with a small screen showing:
**"BOARD_RAMPS_14_EFB
1280 or 2560 CPU?
...CONFIRM
...CANCEL"**
Maybe this is because no actual board is attached to the computer's ports?  
I just hit CONFIRM to see how it went and got an error message:
"LibraryManager: Installing SailfishLCD

Error: Please install Git client from https://git-scm.com/downloads
"

Bob-the-Kuhn, 2019-06-08:
I wish they'd stock the boards in Hong Kong.  That saves a week (at least) by skipping China customs.

Please ZIP & post your configuration.h, configuration_adv.h & pins_YOUR_BOARD.h files.  I'll see what happens.  This assumes that these are the files you want to build with for your new board.

bwaslo, 2019-06-08:
Here they are:

[pins_BIGTREE_SKR_V1.3.zip](https://github.com/MarlinFirmware/MarlinDocumentation/files/3268906/pins_BIGTREE_SKR_V1.3.zip)

bwaslo, 2019-06-08:
It usually takes about the same time or longer to get stuff from Hong-Kong to here than from China.

Bob-the-Kuhn, 2019-06-09:
I couldn't reproduce your problem so I re-read your message.  

Looks like I need to add a section about installing  Git client.

Bob-the-Kuhn, 2019-06-09:
Did it build after installing Git?

bwaslo, 2019-06-09:
Hi Bob,

Yes, I tried it this morning (for the Atmega 2560) and it went through
successfully.  Git seems to be what was needed (though I still have little
idea of what Git actually is!  It's docs never seem to get around to
exactly what it's for).
Thanks for your help!
BTW. where is the information stored/selected for the kind of board it is
to compile for?

Bill

bwaslo, 2019-06-09:
I found and downloaded the version of Marlin 2.0 referenced by the manufacturer at [https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3](url)
But when I tried to do an AutoBuild with that, it wouldn't work, gave me the following error message:
___________
**Process Palette: PIO Build
Executed : python buildroot/share/atom/auto_build.py build
Returned with code 1
'python' is not recognized as an internal or external command,
operable program or batch file.**
___________
(sorry to be  dummy, not familiar with Python at all)

Bob-the-Kuhn, 2019-06-09:
Git is a tool set that allows multiple users to access and update a database of source code.  The database and tool set implement a system that tracks and controls the changes.  Marlin is one of the many databases on the github.com site.

PlatformIO uses a recipe to control how a board is compiled.  The recipe top levels used by Marlin are in the file **platformio.ini**.  The Auto Build script uses info in the file **pins.h** to select the recipe based on the board specified in the file **configuration.h.**

Bob-the-Kuhn, 2019-06-09:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3 is an offshoot of Marlin.  It is not setup to use the Auto Build script.  You'll need to use the vendor's instructions on compiling.

All the software for the Marlin boards is based on the Arduino model.  In the beginning there was the Arduino IDE but now newer systems like PlatformIO are available to compile the software.  The Arduino IDE is the most popular system in general.  PlatformIO and the other Arduino IDE alternatives all have relative strengths and weakneses.

bwaslo, 2019-06-09:
Thanks again.  I've used the Arduino IDE (for Marlin 1.16).  The Biqu instructions talk about using Visual Studio (and PlatformIO), so I should go that way rather than with Atom?

I was reading through some docs and ran into a note that Atom should be started from a console line rather than an icon (I'm in Win10 64).  I tried that (actually, just typed atom in the "Type here to search" box at bottom of Win10 screeen) and ran that way and it compiled.   Shows [Success] in the tk screen (though there was a lot of yellow flying by during the compile).

bwaslo, 2019-06-09:
OK, never mind about the above -- I ran from the icon and it still works now.

If it matters, I had installed a package called "script" as when I searched 'python' in Atom "Install" it said to "use script instead please".  But I uninstalled that and compile still works.  Maybe the magic "exit the app and restart" did it? :)

Bob-the-Kuhn, 2019-06-09:
Lots of different ways to get the same thing done.

Personally I only use Atom as a convenient way to use PlatformIO and the Auto Build script.  I do all my editing in Notepad++.  The big boys like Visual Studio and Sublime.  I never found a convenient method of launching Auto Build from Visual Studio that could be easily shared with the Marlin community.

closed 2019-06-11 by bwaslo

thinkyhead, 2019-06-14:
I prefer "Visual Studio Code" over "Visual Studio" — mainly just cuz it's free.