MarlinFirmware/MarlinDocumentation #226: Platformio Mac install

keithellis74, 2019-06-17:
Can we have some instructions for installing a working instance of Platformio on Mac OS 10.14.5 please.  Tk inter is not installed with this version Mac OS.  I have tried installing tkinter via https://platform.activestate.com/keithellis/ActiveTcl-8.6/distributions?new_project=true, but it is not picked up by platformio.
I get the following error when selecting AutoBuild/PIO Build.
thank you.

thinkyhead, 2019-07-05:
Yeah, I got stuck there too. It seemed to resolve itself, but I'm not sure why. As I recall I may have enabled an option in Visual Studio Code related to the shell environment, but I can't remember which one. If I can find that info I'll let you know.

thinkyhead, 2019-07-05:
@charlesmike hello

shitcreek, 2019-07-06:
[image]

Confirm working after granting permission to modify shell, doesn't come to front.

shitcreek, 2019-07-09:
These instructions worked for me on Mac: http://marlinfw.org/docs/basics/install_platformio.html

closed 2019-07-09 by shitcreek