MarlinFirmware/MarlinDocumentation #233: [Bug] Online image converter buggy

VanessaE, 2019-07-07:
~~Foreword:  I'm putting this here in the Marlin issue tracker because I don't see a more appropriate place.~~ Issue moved here at @Ludy87's suggestion.

### Description

The online image converter located at http://marlinfw.org/tools/u8glib/converter.html has several minor bugs, when generating a status screen logo:

1. The format/syntax of code block it creates is not Marlin-2.0 compatible:
  * The converter sets the image width as `STATUS_SCREENWIDTH`, but Marlin 2.x expects `STATUS_LOGO_WIDTH`.
  * The name of the array (?  what we call a table in Lua) is incorrect.  The converter sets it to 
`status_screen0_bmp` but Marlin expects `status_logo_bmp`.
  * Minor nitpick:  the converter is appending a redundant comma to the end of the bitmap data.
  * Not sure if it matters:  `#pragma once` is used in other example status screen images supplied with Marlin, but the converter does not insert that directive into its output.

~~2. The converter is inserting at least one space between items in the last line of output, where it should be inserting a comma.  This confuses the compiler, causing it to error out:
`  B00000000,B00000000,B00000000,B00000001 B11000000,`
(look between the 4th and 5th bytes)~~ (already addressed by @Ludy87 in https://github.com/Ludy87/MarlinDocumentation/commit/adbccd23dcbd0e49a7d28c997ae012844c22ad6d)

3. On the converter page, the text "If this tool doesn't work in your browser, use the server-side tool at digole.com." links (at the end there) to a page that declares itself as deprecated, which then recommends users follow another link to an "online emulator", which, to put it politely, looks like something straight out of 1990, and seems not to be particularly useful...  I recommend just deleting this line of text from the page.

### Steps to Reproduce

Browse to http://marlinfw.org/tools/u8glib/converter.html and plug in a suitable image file, examine the output.

**Expected behavior:**

I should get a block of code I can copy/paste directly into _Statusscreen.h

**Actual behavior:**

I got a block block of code that required an enormous :wink: 4-line change to make it work.

#### Additional Information
<details><summary>_Statusscreen.h as generated by the converter</summary>

```
/**
 * Made with Marlin Bitmap Converter
 * http://marlinfw.org/tools/u8glib/converter.html
 *
 * This bitmap from the file 'marlin-prusa-i3-status-screen-logo.png'
 */
#define STATUS_SCREENWIDTH 40
const unsigned char status_screen0_bmp[] PROGMEM = {
  B00000001,B11111111,B00000001,B00000000,B00000000,
  B00000010,B00000111,B00000011,B00000000,B00000000,
  B00000100,B01010011,B00000011,B00000000,B00000000,
  B00000100,B10101001,B00000111,B10000000,B00000000,
  B00000100,B10101001,B00000111,B10000000,B00000000,
  B00000100,B10101001,B00000111,B11001110,B00000000,
  B00000100,B10001001,B00001000,B01111010,B00000000,
  B00000010,B00000010,B00001111,B10010100,B00000000,
  B00000001,B11111100,B00011111,B11111100,B00000000,
  B00000000,B00000000,B00111111,B11111110,B00000000,
  B00000000,B00000001,B11111111,B11111111,B00000000,
  B00000011,B11111111,B11110111,B11111111,B10000000,
  B00000000,B00001111,B10111011,B11111111,B10000000,
  B00000000,B00000001,B11111011,B11101101,B11000000,
  B00000000,B00000010,B11111011,B11110101,B11000000,
  B00000000,B00000101,B10110111,B11110101,B11000000,
  B00000110,B00001110,B00001111,B11111110,B11000000,
  B00000101,B00000000,B00000000,B00000110,B11000000,
  B00000101,B01010101,B00110011,B10000110,B11000000,
  B00000110,B01100101,B01100100,B10000011,B11100000,
  B00000100,B01000101,B00110101,B10000010,B11100000,
  B00000100,B01000011,B01100010,B10000001,B11110000,
  B00000000,B00000000,B00000000,B00000001,B11010000,
  B00000000,B00001110,B00000000,B00000011,B10000000,
  B00000000,B00100010,B00000000,B00000111,B10000000,
  B00000000,B00000100,B00000001,B11111111,B10000000,
  B00000000,B00100010,B00000111,B11111100,B10000000,
  B00000000,B00100010,B00001100,B00000000,B11000000,
  B00000000,B00101100,B00000000,B00000111,B11000000,
  B00000000,B00000000,B00000000,B00000001 B11000000,
};
```
</details>

<details><summary>The corrected code</summary>

```
/**
 * Made with Marlin Bitmap Converter
 * http://marlinfw.org/tools/u8glib/converter.html
 *
 * This bitmap from the file 'marlin-prusa-i3-status-screen-logo.png'
 */

#pragma once

#define STATUS_LOGO_WIDTH 39

const unsigned char status_logo_bmp[] PROGMEM = {
  B00000001,B11111111,B00000001,B00000000,B00000000,
  B00000010,B00000111,B00000011,B00000000,B00000000,
  B00000100,B01010011,B00000011,B00000000,B00000000,
  B00000100,B10101001,B00000111,B10000000,B00000000,
  B00000100,B10101001,B00000111,B10000000,B00000000,
  B00000100,B10101001,B00000111,B11001110,B00000000,
  B00000100,B10001001,B00001000,B01111010,B00000000,
  B00000010,B00000010,B00001111,B10010100,B00000000,
  B00000001,B11111100,B00011111,B11111100,B00000000,
  B00000000,B00000000,B00111111,B11111110,B00000000,
  B00000000,B00000001,B11111111,B11111111,B00000000,
  B00000011,B11111111,B11110111,B11111111,B10000000,
  B00000000,B00001111,B10111011,B11111111,B10000000,
  B00000000,B00000001,B11111011,B11101101,B11000000,
  B00000000,B00000010,B11111011,B11110101,B11000000,
  B00000000,B00000101,B10110111,B11110101,B11000000,
  B00000110,B00001110,B00001111,B11111110,B11000000,
  B00000101,B00000000,B00000000,B00000110,B11000000,
  B00000101,B01010101,B00110011,B10000110,B11000000,
  B00000110,B01100101,B01100100,B10000011,B11100000,
  B00000100,B01000101,B00110101,B10000010,B11100000,
  B00000100,B01000011,B01100010,B10000001,B11110000,
  B00000000,B00000000,B00000000,B00000001,B11010000,
  B00000000,B00001110,B00000000,B00000011,B10000000,
  B00000000,B00100010,B00000000,B00000111,B10000000,
  B00000000,B00000100,B00000001,B11111111,B10000000,
  B00000000,B00100010,B00000111,B11111100,B10000000,
  B00000000,B00100010,B00001100,B00000000,B11000000,
  B00000000,B00101100,B00000000,B00000111,B11000000,
  B00000000,B00000000,B00000000,B00000001,B11000000
};
```
</details>

(note: I'm not sure whether it was really necessary to set the width to 39 pixels, but that's what one of the other stock Marlin images used, and I assumed a value one pixel less than a whole byte's worth may be important, so I followed suit)

Incidentally, here is the image I used, scaled up for visibility in this page:
[image]

original size:
[image]

(If you're interested in including that image/its code in Marlin as an example for i3's, feel free to do so.  It's my own work, though obviously derived from the original Marlin logo, so whatever license the original uses, or CC-By-SA if you like)

thinkyhead, 2019-07-08:
It's on the TODO list, but so many other things TODO…

VanessaE, 2019-07-08:
Fair enough. :smiley:

benebrady, 2019-07-10:
The state of the bitmap converter is not even Marlin 1.1.x compatible at this point. An extra byte of Zero is added to each line in the file.  This zip file contains 2 bootscreens. Both were generated by the bitmap converter. One was edited to remove the extra byte of Zero at the end of each line. The edited file works under 8 bit controllers such as the MKS Gen L and MKS Gen 1.4, but it DOES not work for SKR 1.3 boards. The NON-edited bootscreen file also does not work in the 8 bit controllers.
[_Bootscreen.zip](https://github.com/MarlinFirmware/MarlinDocumentation/files/3375279/_Bootscreen.zip)

thinkyhead, 2019-07-11:
The author of the tool is aware of its shortcomings.

VanessaE, 2019-08-09:
@shitcreek why did you close this?

shitcreek, 2019-08-09:
Sorry I didn't see the checklist and thought it was all resolved.

LMF5000, 2019-08-17:
Can confirm bug - this does not work on SKR E3 because it's adding an extra B00000000 entry at the end of each line. If you count the number of entries, you'll see there's 17 [Bxxxxxxxx] entries per line, instead of 16. The result is a garbled image. The interim solution is to open the output file with notepad++ and delete the last entry on each line, or run a find and replace, finding "B00000000\r\n" and replacing with "\r\n". Then manually delete the very last B00000000 at the bottom of the file.

shitcreek, 2019-09-28:
I've corrected the code to address the remaining issues and the extra column. Now awaiting site rebuild @thinkyhead

closed 2019-10-22 by thinkyhead