MarlinFirmware/MarlinDocumentation #236: Origin of Linear Advance

BagelOrb, 2019-08-14:
@Sebastianv650 
I would like to know the origins of the linear advance method so that I can cite the proper sources in my journal publication.

Is the algorithm based on any published paper?

If not, would this documentation be the proper citing source?

Sebastianv650, 2019-08-14:
Hello @BagelOrb, I'm curently on vacation therefore if you need detailed information I could provide them in about 1,5 weeks. Typing longer text on a mobile phone isn't fun ;)

In short, the most important (maybe?) information I can give you is that there is a publication waiting for release just now, you can find it in pre-release state (EarlyCite) on emerald publishing. It should be in the Rapid Prototyping Journal soon.
Regarding the origin, LA isn't based on a published paper. I developed it based on my finding that the nozzle pressure isn't right on line starts and ends. During my first research, I found the JKN algorithm and some (sparse) notes about it in the internet. There was also an "advance" function in Marlin back then, but when I startet coding LA this one was already not supported any more and the code was broken. Therefore I started my own implementation from scratch. But I would say JKN had an influence as a first inspiration.

BagelOrb, 2019-08-14:
I'm just looking for a good reference I could cite which explains what it is and why.
What is the title of your paper? I will include a citation if I can find it online.

Sebastianv650, 2019-08-15:
For this purpose the Marlin documentation is most likely the best source up to now.
The title is 'Investigating pressure advance algorithms for filament-based melt extrusion additive manufacturing: theory, practice and simulations '.

BagelOrb, 2019-08-15:
Thanks a bunch, Mr Popp!

You'll have at least one citation in a year or so :D

Enjoy your vacation!

closed 2019-08-15 by BagelOrb