MarlinFirmware/MarlinDocumentation #238: Page does not describe the function of #define MANUAL_PROBE_START_Z

LMF5000, 2019-08-17:
Neither this page, nor configuration.h describe the function of the setting MANUAL_PROBE_START_Z 0.2

From my experiments, it appears the functionality is as follows:
Sets the z-height the printer initially moves to at each mesh point during manual probing. Useful to prevent the nozzle colliding with the bed if that point happens to be warped upwards. The disadvantage is that the knob will have to be turned further for each mesh point in case of a perfectly flat bed where each point is at Z=0.

shitcreek, 2019-09-12:
Added and credited you, now awaiting site rebuild.

closed 2019-09-12 by shitcreek

shitcreek, 2019-09-13:
It would make sense that it's the latter: using the saved z positions from the previous run.
This is how I am wording it, feel free to tweak:

> `MANUAL_PROBE_START_Z` sets the z-height the printer initially moves to at each mesh point during manual probing. With this disabled, the printer moves to the Z position of 0 for the first probe point. Each followig probe point uses the Z position of the probe point before it.

LMF5000, 2019-09-13:
How about this wording:

MANUAL_PROBE_START_Z sets the z-height the printer initially moves to at each mesh point during manual probing. With this disabled, the printer moves to Z=0 for the first probe point, then Z=last point z for each subsequent probe point.

shitcreek, 2019-09-14:
Kind of confusing at the end there.
How about:
...With this disabled, the printer will move to Z0 for the first probe point, then each consecutive probe point uses the Z position of the point preceding it.

LMF5000, 2019-09-14:
Sounds perfect :)