MarlinFirmware/MarlinDocumentation #266: marlinfw.org is displaying a 404 page

Ghostkeeper, 2019-12-25:
I'm not sure if this is the correct place to notify you of this, but I can't find any other. Maybe you've already seen this, too.

The website www.marlinfw.org currently displays a 404 page, saying that the Github Pages website can't be found.

[image]

I'm seeing some recent work on the deployment logic in the history of this repository. Perhaps this has something to do with it?

Ghostkeeper, 2019-12-25:
Oops, seems someone else was reporting this at the same time. I'm sorry. Please continue discussion here: https://github.com/MarlinFirmware/MarlinDocumentation/issues/265

closed 2019-12-25 by Ghostkeeper

thinkyhead, 2020-01-19:
We tend to notice this, so no need to report it unless it persists for days.