MarlinFirmware/MarlinDocumentation #267: Documentation

p2rjp0, 2019-12-27:
Changes to Probe Offset incorrect in documentation.
Needs block above MIN_PROBE_EDGE with following.
/**
 * Z Probe to nozzle (X,Y) offset, relative to (0, 0).
 *
 * In the following example the X and Y offsets are both positive:
 *
 *   #define NOZZLE_TO_PROBE_OFFSET { 10, 10, 0 }
 *
 *     +-- BACK ---+
 *     |           |
 *   L |    (+) P  | R <-- probe (20,20) 
 *   E |           | I
 *   F | (-) N (+) | G <-- nozzle (10,10)
 *   T |           | H
 *     |    (-)    | T
 *     |           |
 *     O-- FRONT --+
 *   (0,0)
 *
 * Specify a Probe position as { X, Y, Z }
 */
#define NOZZLE_TO_PROBE_OFFSET { 20, 20, 0 }

thinkyhead, 2020-01-19:
Thanks for the suggestion. I'm not sure it is really needed and it adds visual clutter.

closed 2020-01-19 by thinkyhead

p2rjp0, 2020-01-20:
Sorry, for full amount.  The Key to what I was pointing out was that the Offset became an array rather than three separate values.