MarlinFirmware/MarlinDocumentation #270: Installing Installing Marlin (VSCode) is missing installation of git

PlasmaHH, 2020-01-26:
I tried to follow the information step by step and stumbled over the part to clone a git repo: vscode needs an installation of git itself for this to work (https://stackoverflow.com/a/46620901)

thinkyhead, 2020-02-15:
There is no reason to use VSCode to manage your Marlin git repository. Use GitHub Desktop instead.

closed 2020-02-15 by thinkyhead

PlasmaHH, 2020-02-18:
I think thats kinda besides the point. The marlin websites states that vscode is the preferred way to go, then gives a step by step instruction to install vscode and clone the git directory which is missing that vital step for people that never installed it before.

thinkyhead, 2020-02-22:
Please submit a re-write that removes references to git. You only need to download a ZIP of Marlin. No git is required.