MarlinFirmware/MarlinDocumentation #274: Where to acess?

lukeskymuh, 2020-03-06:
Hi,

it is great that there is this project. I am just confused where to best acess this documentation. Can you add instructions or link to the project description how to best acess this documentation (for non github experts)? Is it hosted by a web site, do I have to download the files locally?

BR luke

filipgoc, 2020-03-06:
Is this what you're looking for? 

¯\_(ツ)_/¯

[image]

lukeskymuh, 2020-03-06:
So the Homepage is the project. for me this was not very clear from the description. A sentence in the description would be usefull. Specially when accessing the marlinfw.org with a mobile device.  I mainly could see only generic information or topics I am not interested in. which was  not very usefull. With a PC I can see that there is much more content and it is well structured.

Unfortunatly the topic I am looking for is not included yet: How to implement new kinematics. It is only mentioned that different kinematics are implemented. 
If it does not exist already elsewere I will dig into the code and write a step by step procedure for other kinematics and contribute to the project. If you can give some hints this would be very usefull.

filipgoc, 2020-03-14:
This kind of info would be beyond the scope of the docs - these are docs for users mostly, not really for developers. As a developer, the best I can recommend is fulltext search and read the source code - it's actually well annotated in general. Also, for this kind of stuff you want to be hanging out in the Marlin repo, not here. 

https://github.com/MarlinFirmware/Marlin

Read the readme. There's various things from coding standards to where you can go and talk about these things (discord server etc.) Good luck!

lukeskymuh, 2020-03-16:
Thank you. I will check it. In order to avoid those question to be asked to many times at the wrong place, a section where to go for new developments and special addaptations.

thinkyhead, 2020-03-31:
Here are some helpful resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-03-31 by thinkyhead