MarlinFirmware/MarlinDocumentation #276: Linear Advance Calibration Generates an Unnecessary Move

gayuha, 2020-03-12:
When I generate the calibration pattern, this line is generated near the beginning:

`G1 X100 Y100 F9000 ; move to start`

This moves the nozzle extremely fast and close to the bed, and risks scratching it.

thinkyhead, 2020-03-31:
I've added a 5mm raise just after G28 and before heating. That should do the trick.

closed 2020-03-31 by thinkyhead

gayuha, 2020-04-01:
That is indeed better, but I thought this line should simply be removed. I remove it manually before printing and it works ok. Is there a purpose for it?

thinkyhead, 2020-04-12:
More a question for the tool's author, @Sineos.

Sineos, 2020-04-14:
The ; move to start lines are generated whenever we need to move to a new starting point without printing. The speed is determined with the Movement Speed setting in the tool.

gayuha, 2020-04-20:
I refer specifically to the first ` ; move to start` line that is generated. It just moves the nozzle to the center of the bed. No printing happens there.