MarlinFirmware/MarlinDocumentation #283: Tool Change T0 , T1  (Extruder Change) not listed

firepower9966, 2020-03-14:
T0 and T1, etc  are used by Cura for changing active extruder
but its not listed in Marlin GCode listing.
Can it be added especially for people new to GCode who may want to know what it is.

shitcreek, 2020-04-10:
I've just created these here: https://github.com/MarlinFirmware/MarlinDocumentation/pull/290
I'm not sure what all to have it say though. Any recommendation?

firepower9966, 2020-04-10:
I would include;  switch to tool head / hotend or* virtual* tool head
hotend (for example mixing hot end with 2 phyisical extruder motors can
have up yo 16 virtual extruders with different mix ratios.)
important to get it online and it can be edited and updated later.

thank you.

shitcreek, 2020-04-14:
Okay, thanks, it's up now. I've added your bit. Need to wait for site rebuild to see the new changes.

closed 2020-04-14 by shitcreek