MarlinFirmware/MarlinDocumentation #288: K-factor Calibration Pattern generator not working

wcyb, 2020-04-01:
Looks like in the file: MarlinDocumentation/_tools/lin_advance/k-factor.js in line 60 is a bug.
Instead of `BED_LEVELING = SELECT_BED_LEVELING.val(),`
it should be `BED_LEVELING = $('#SELECT_BED_LEVELING').val(),`

xyz8858, 2020-04-01:
I logged in just to flag this. Glad there is already activity. As wcyb said, there is a code problem. :)

ProfEngr, 2020-04-01:
I assume this is keeping the Generate button from working?

adwellj, 2020-04-01:
@ProfEngr, yes, that's why it's not working. In the meantime, you can get it to work by editing k-factor.js in DevTools on line 60. Change it from `SELECT_BED_LEVELING.val()` to `SELECT_BED_LEVELING.value` and then type Ctrl+S to save the changes locally. This worked for me in Chrome; I would assume other browsers could be similar.

gaskins1992, 2020-04-01:
@adwellj, Thanks for the solution. For me, this worked for Chrome, but not Safari.

closed 2020-04-01 by thinkyhead