MarlinFirmware/MarlinDocumentation #303: invert Endstop signal.

jonawald, 2020-05-02:
Hello. I'm not sure if this is an issue with Marlan or an issue with my google ability, but I have been searching for a few hours now on how to invert signal from an endstop. My printer was on Repetier before this. 

My z-probe is inverted. It is triggered when not touching the bed and open when in contact with the bed. In repetier it was an easy fix. I inverted the signal from that endstop. I cannot find out where to do this in Marlan.

Jonathan

AnHardt, 2020-05-02:
How about
`#define Z_MIN_PROBE_ENDSTOP_INVERTING false // Set to true to invert the logic of the probe.
`

jonawald, 2020-05-02:
Well. I guess I missed that somehow. I thought it must be there, but I just could not find it. Is that on configuration.h?

jonawald, 2020-05-03:
Thank you AnHardt. It was right there, but I missed it somehow.

jonawald, 2020-05-03:
But after trying back and forth a few times, it does not make a difference for me. Still inverted from what I need. I have just two wires going to switch, one to Vcc the other to Signal. 
`#endif
//===========================================================================
//============================== Endstop Settings ===========================
//===========================================================================

// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
//#define USE_XMIN_PLUG
//#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG // a Z probe
#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
#define USE_ZMAX_PLUG

// Enable pullup for all endstops to prevent a floating state
//#define ENDSTOPPULLUPS
#if DISABLED(ENDSTOPPULLUPS)
  // Disable ENDSTOPPULLUPS to set pullups individually
  #define ENDSTOPPULLUP_XMAX
  #define ENDSTOPPULLUP_YMAX
  #define ENDSTOPPULLUP_ZMAX
  //#define ENDSTOPPULLUP_XMIN
  //#define ENDSTOPPULLUP_YMIN
  #define ENDSTOPPULLUP_ZMIN
  #define ENDSTOPPULLUP_ZMIN_PROBE
#endif

// Enable pulldown for all endstops to prevent a floating state
//#define ENDSTOPPULLDOWNS
#if DISABLED(ENDSTOPPULLDOWNS)
  // Disable ENDSTOPPULLDOWNS to set pulldowns individually
  //#define ENDSTOPPULLDOWN_XMAX
  //#define ENDSTOPPULLDOWN_YMAX
  //#define ENDSTOPPULLDOWN_ZMAX
  //#define ENDSTOPPULLDOWN_XMIN
  //#define ENDSTOPPULLDOWN_YMIN
  //#define ENDSTOPPULLDOWN_ZMIN
  //#define ENDSTOPPULLDOWN_ZMIN_PROBE
#endif

// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
#define X_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING true  // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe.
`

I think I'm using the correct setting for resistor because I'm getting a change in signal when I trigger it, but How do I get it to actually invert?

jonawald, 2020-05-03:
Ok, that was my mistake again. Sorry. I changed this
#define Z_MIN_PROBE_ENDSTOP_INVERTING true // Set to true to invert the logic of the probe.

instead of this.
 #define Z_MIN_ENDSTOP_INVERTING false // Set to true to invert the logic of the endstop.

shitcreek, 2020-05-03:
Glad you've figured it out. Next time please try the discord channel or facebook page for tech support as this queue is for issues relating to the documentation of Marlin. Thanks.

closed 2020-05-03 by shitcreek

jonawald, 2020-05-03:
One more question please. Where do I find links to your discord channel or facebook page? I can't find on Marlin homepage or with a quick google search.

shitcreek, 2020-05-03:
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y