MarlinFirmware/MarlinDocumentation #310: Probes documentation outdated

lnlp, 2020-06-02:
I ran into an issue with BLTouch so I consulted the documentation and ran into the following issue:

Below documentation seems outdated because no longer supported  in the code.

On https://marlinfw.org/docs/configuration/probes.html

> If the probe is connected to any other pin, enable Z_MIN_PROBE_ENDSTOP.

In code (2.0.5.3) SanityCheck.h:
```cpp
#elif defined(Z_MIN_PROBE_ENDSTOP)
  #error "Z_MIN_PROBE_ENDSTOP is no longer required. Please remove it from Configuration.h."
```

thinkyhead, 2020-06-03:
Oh yes, so much is out of date, and so few contributors. If you have the time, please submit corrections to any outdated things you see.

shitcreek, 2020-06-10:
I've added a note for this. Thanks.

closed 2020-06-10 by shitcreek

lnlp, 2020-06-10:
Reported issues shall not be closed when they have not been fixed.
The documentation page however still mentions to use Z_MIN_PROBE_ENDSTOP.

Making a note of a problem does not solve that problem.
Therefore this issue should be reopened again.

_Closing reported issues while they have not been fixed will render the GitHub Issues mechanism completely useles. It will also not motivate people to report issues and is also prone to reporting the same issue multiple times.
Issues are meant to give insight in which are already known and have already been reported and whether an issue is still present (open) or has been fixed already (closed).