MarlinFirmware/MarlinDocumentation #314: Just where on the 20x4 char display is there room for a chamber temp?

marcomortale, 2020-06-17:
Just where on the 20x4 char display is there room for a chamber temp?
Each temperature displayed uses 10 characters. A icon + 3 char current temp + / + 3 char set temp + degrees symbol + space separator.  
So with a hotend and a heated bed, you have no space to put any further temp displays.

_Originally posted by @ellensp in https://github.com/MarlinFirmware/Marlin/issues/18338#issuecomment-645324998_

closed 2020-06-17 by marcomortale

karabas2011, 2020-06-18:
How do you see temp on two nozzles setups? 
Replacing X Y output as less impotant. 
What the problem?

thinkyhead, 2020-06-21:
When space runs out then you could start to cycle what is displayed in the last position. It would also be possible to design a whole new and more compact layout. Going forward, DWIN screen support is improving, so there will be more pixels and colors to work with.