MarlinFirmware/MarlinDocumentation #320: M281 gcode not working in pronterface

ReddyVidya0398, 2020-07-04:
I am facing a small problem with the firmware.Today I tried testing the editable servo angles feature in configuration.h.Tried M281 P0 L0 U90 in pronterface.It returns unknown command.But when I try M280 P0 S0 it works perfectly.Any suggestions?
I also tried M281 P0 L90 U0.I tried M502 followed by M500 and M501 still same response.
SENDING:M281 P0 L90 U0
echo:Unknown command: "M281 P0 L90 U0"

ellensp, 2020-07-04:
not related to documentation.  Is just a config file issue, as explained in discord.
To enable M281 you need to have EDITABLE_SERVO_ANGLES and one of these set 
Z_PROBE_SERVO_NR, SWITCHING_EXTRUDER, SWITCHING_NOZZLE ie something that has a servo.

ReddyVidya0398, 2020-07-04:
But is there anyway I can set the servo upper and lower limits without Z_PROBE_SERVO_NR, SWITCHING_EXTRUDER, SWITCHING_NOZZLE options enabled? Whice file in the source code will I need to make modifications in?

thinkyhead, 2020-07-05:
M281 is for servos attached to a "deploy and stow" type of device where a servo is treated kind of like a solenoid, with only two positions. For regular servo usage, just set `NUM_SERVOS` and use `M280` to set a specific angle for your application. There is no concept of servo-emulating-solenoid outside of probe-like devices.

closed 2020-07-05 by thinkyhead