MarlinFirmware/MarlinDocumentation #341: Missing documentation for various commands

costas-basdekis, 2020-10-04:
I'm using the excellent documentation [as a plugin for Octoprint](https://plugins.octoprint.org/plugins/marlingcodedocumentation/), and in the [parser](https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/gcode/parser.cpp#L245) there is support for commands M810-M19, and in [gcode/feature/macro/M810-M819.cpp](https://github.com/MarlinFirmware/Marlin/blob/92076c65601de5fdfe0facf32dd60c57aab05747/Marlin/src/gcode/feature/macro/M810-M819.cpp#L37) you can see:

```
/**
 * M810_819: Set/execute a G-code macro.
 *
 * Usage:
 *   M810 <command>|...   Set Macro 0 to the given commands, separated by the pipe character
 *   M810                 Execute Macro 0
 */
```

But there is no entry in this repo about those commands.

Furthermore, I checked to see if there are any more such commands, and I found 43 commands in total (including the above) that are not documented:
`G6, G38, G800`
`M12, M876, M876, M202, M305, M430, M510, M511, M512, M672, M810, M811, M812, M813, M814, M815, M816, M817, M818, M819, M916, M917, M918, M800, M860, M861, M862, M863, M864, M865, M866, M867, M868, M869, M951, M993, M994, M995, M1000, M1001`

I also checked for commands that are documented, but are don't seem to be referenced in the code, and there's only one:
`M915`

I'm not sure how many people are using these, and whether it's worth the effort, but I thought I'd bring it to your attention, since this repo is an excellent source of structured documentation.

thinkyhead, 2020-10-06:
Thanks! If you find any holes in the documentation and if you have the time, please feel free to fill them in. The documentation is also open and free for anyone to contribute, and this is just the kind of thing that gets missed due to a shortage of attentive maintainers.

costas-basdekis, 2020-10-06:
I might do that.

Although, on cases like `G6`, I can't find much documentation except [some explanation in configuration](https://github.com/MarlinFirmware/Marlin/pull/17853/files#diff-18dab9d337cee2f9d2209940ed08dbacR1666-R1674), and [RepRap wiki](https://reprap.org/wiki/Direct_Stepping), which I guess are enough.

What other sources do you usually use to fill documentation?

thinkyhead, 2020-10-06:
> What other sources do you usually use to fill documentation?

For G6 our best bet is to just give an overview of the parameters as described in `G6.cpp` and then supply links to the project. As more information becomes available we can add it in, and when inevitably someone does a video on the subject we can add the video ID to the page too.

And for other G-codes, I will first pull up any documentation I (or someone else) already added to the RepRap wiki for reference, then I'll build the new page using that plus the description of the G-code in its .cpp file. If it seems like parameter descriptions are missing there, I will scan the G-code to see if `parser.seen` refers to anything else, then fix up the descriptions in the code and make sure to include all parameters in the website.

I try to make a pass at the documentation at least once a week, but truly I am at the whim of the issue queue….

Speaka, 2020-10-08:
I added a doc file for M810-819 hopefully it is all right. Since i tried to use macros after i saw your issue i thought this would be helpful for other people browsing the doc. If more pages are wanted a may add some rudimentary descriptions. At least every GCode should appear in the docs whether its good documented or not.

costas-basdekis, 2020-10-08:
You can find them in [parse.cpp](https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/gcode/gcode.cpp#L879-L890) and in [encoder_i2c.cpp](https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/feature/encoder_i2c.cpp#L807)

shitcreek, 2020-10-09:
added `G6` doc - tagging @colinrgodsey for his input (site needs rebuild before the page will show up - @thinkyhead)

Speaka, 2020-10-09:
**M202** is not used:
`      #if 0
        case 202: M202(); break;                                  // M202: Not used for Sprinter/grbl gen6
      #endif`
Can't this also be deleted from [gcode.cpp](https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/gcode/gcode.cpp#L591-L593)?

shitcreek, 2020-10-09:
@Speaka I agree that comment is not relevant to Marlin.

added M860 - M869 (awaiting site rebuild)

shitcreek, 2020-10-16:
Alright, I've went through the list and added them:
G6
G38 - it’s already there but under G38.2, G38.3, G38.4, G38.5
G800, M800 - same diff
M202 - this Gcode doesn’t exist
M305
M430
M510, M511, M512
M672
M860-M869
M876
M916, M917, M918
M951
M993, M994
M995
M1000
M1001

------
awaiting site rebuild.

Speaka, 2020-10-17:
Looking good so far. I think for every non default option which requires special include definitions there should be a hint on the page. E.g 
G6: notes:  - Requires `DIRECT_STEPPING`
M510: PASSWORD_FEATURE
M860: I2C_POSITION_ENCODERS
M916-918: MONITOR_L6470_DRIVER_STATUS

shitcreek, 2020-10-20:
Only `G6` was missing the requirement note. The rest has it in two places.
I think it is fair to close this now that we've added the missing gcode docs.

closed 2020-10-20 by shitcreek

thinkyhead, 2020-10-21:
Be sure to check that codes have also been added to the RepRap wiki. I made sure to add the `M510`-`M512` codes yesterday, but haven't checked others lately.