MarlinFirmware/MarlinDocumentation #347: marlin2.07.0.2

mboccus2612, 2020-11-09:
When start printing the z Nozzle head rise up 100mm and back to the print  and even on the first layer it rise each time to print something, is it in configuration or adv configuration.h that i miss somethiong or a bug, video can folloow if u request.

ellensp, 2020-11-09:
How is this a documentation issue?

shitcreek, 2020-11-17:
This is an issue queue for Marlin Documentation.

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-11-17 by shitcreek