MarlinFirmware/MarlinDocumentation #350: M305 options incorrect?

VanessaE, 2020-12-06:
The page documenting the `M305` command (https://marlinfw.org/docs/gcode/M305.html) has a minor error.  The bottom of the page reads:

[image]

The example command at the bottom there conflicts with the options list just above -- I think you meant to write `[C<coeff>]  Steinhart-Hart Coefficient ‘C’` in the options list.

thisiskeithb, 2020-12-19:
> I think you meant to write [C<coeff>] Steinhart-Hart Coefficient ‘C’ in the options list.

Confirmed. This matches [M305.cpp](https://github.com/MarlinFirmware/Marlin/blob/45b11553f40c48f03bfa10b542f27a9555cb2583/Marlin/src/gcode/config/M305.cpp#L69-L71):
```cpp
    if (parser.seen('C')) // Steinhart-Hart C coefficient
      if (!thermalManager.set_sh_coeff(t_index, parser.value_float()))
        SERIAL_ECHO_MSG("!Invalid Steinhart-Hart C coeff. (-0.01 < C < +0.01)");
```
See https://github.com/MarlinFirmware/MarlinDocumentation/pull/357 for the updated documentation.

qwewer0, 2021-01-01:
This can be closed, as the PR has merged.

closed 2021-01-01 by thisiskeithb