MarlinFirmware/MarlinDocumentation #354: G34 tolerance allowed in 0.01 not 0.001

bakaufman, 2020-12-17:
Although you can compile to have tolerances for G34 to be T0.001, you cannot run it. A message shows up saying 0.01-1. The language on the website needs to chang 0.001 to 0.01

thisiskeithb, 2020-12-17:
> The language on the website needs to chang 0.001 to 0.01

Confirmed. This matches [G34_M422.cpp](https://github.com/MarlinFirmware/Marlin/blob/e9677594ea164319dea1c6cc2dc905237ad0dec4/Marlin/src/gcode/calibrate/G34_M422.cpp#L115-L119):
```cpp
      const float z_auto_align_accuracy = parser.floatval('T', Z_STEPPER_ALIGN_ACC);
      if (!WITHIN(z_auto_align_accuracy, 0.01f, 1.0f)) {
        SERIAL_ECHOLNPGM("?(T)arget accuracy out of bounds (0.01-1.0).");
        break;
      }
```
See https://github.com/MarlinFirmware/MarlinDocumentation/pull/355 for the updated documentation.

qwewer0, 2021-01-01:
This can be closed, as the PR has merged.

closed 2021-01-01 by thisiskeithb