MarlinFirmware/MarlinDocumentation #356: MicroSwiss Direct Drive extruder print about 6mm to the left of the print bed

PiEyed-Tom, 2020-12-18:
I know that there is a way to change the X and Y zero point by using the #define X_MIN_POS -6 and #define Y_MIN_POS -10 when creating a new build, but every time I try building a new .bin file, something else doesn't work.  It's probably the "tutorial" I've been using, but I can't figure out how to fix my problem.  Is there a way you could add the changes to the "Firmware Builder (beta)" on your website? That way if anyone wants the current build with BLTouch and MicroSwiss Direct Drive Extruder could "build" there own custom version easily.  X_MINPOS and Y_MIN_POS could have the default X -5 and Y -10 or specify the exact offset their machine requires.

thisiskeithb, 2020-12-18:
> Is there a way you could add the changes to the "Firmware Builder (beta)" on your website?

Marlin doesn't host a firmware builder. Do you have a link to what you're referring to?

PiEyed-Tom, 2020-12-18:
Umm... yes, they do! https://marlin.crc.id.au/builder/

thisiskeithb, 2020-12-18:
> Umm... yes, they do! https://marlin.crc.id.au/builder/

That site is not created/maintained by Marlin. You'll need to contact the site owner.

closed 2020-12-18 by thisiskeithb

PiEyed-Tom, 2020-12-18:
Daily builds of Marlin FW for 3D PrintersBuild your own custom firmware:Not happy with the feature list in the daily built firmware? Select your own options to be included in our custom firmware builder. Use the credits included in your subscription to build your custom feature set, and have your freshly prepared firmware ready in under a minute.OK. Shouldn’t you kame that clear on your mail web page?  It’s the first paragraph at the top of your page. Can you PLEASE tell me who is hosting that web page so that I can contact them?Thanks,Tom Pratt Sent from Mail for Windows 10 From: Keith BennettSent: Friday, December 18, 2020 2:01 PMTo: MarlinFirmware/MarlinDocumentationCc: PiEyed; AuthorSubject: Re: [MarlinFirmware/MarlinDocumentation] MicroSwiss Direct Drive extruder print about 6mm to the left of the print bed (#356) Umm... yes, they do! https://marlin.crc.id.au/builder/That site is not created/maintained by Marlin. You'll need to contact the site owner.

thisiskeithb, 2020-12-18:
> Shouldn’t you kame that clear on your mail web page? 

Again, the link you posted is not owned or operated by Marlin. You'll need to contact them.

PiEyed-Tom, 2020-12-18:
Shouldn’t you kame (sorry, that was) make  that clear on your mail web page?‘You'll need to contact them’Can you tell me who “them” is?I feel like you’re being very rude.  Is that the way Marlin treats their paying customers?  Sent from Mail for Windows 10 From: Keith BennettSent: Friday, December 18, 2020 2:16 PMTo: MarlinFirmware/MarlinDocumentationCc: PiEyed; AuthorSubject: Re: [MarlinFirmware/MarlinDocumentation] MicroSwiss Direct Drive extruder print about 6mm to the left of the print bed (#356) Shouldn’t you kame that clear on your mail web page?Again, the link you posted is not owned or operated by Marlin. You'll need to contact them.

thisiskeithb, 2020-12-18:
> Can you tell me who “them” is?I feel like you’re being very rude.  Is that the way Marlin treats their paying customers? 

Again, that website is not _is not owned or maintained by the Marlin Firmware project or any of its developers_. This is not the proper place to seek support because Marlin is not affiliated with that website.

From that website's homepage, you need to submit bug reports to https://forums.creality3dofficial.com/community/bug-reports/

[image]