MarlinFirmware/MarlinDocumentation #363: RESTORE_LEVELING_AFTER_G28

bakaufman, 2021-01-03:
I was unable to identify language in the MARLINFW CONFIGURATION section that mirrors this section in Configuration.h. The differences in these options are slight and semantic and probably should have more explanation.  
/**
 * Normally G28 leaves leveling disabled on completion. Enable one of
 * these options to restore the prior leveling state or to always enable
 * leveling immediately after G28.
 */
//#define RESTORE_LEVELING_AFTER_G28 
//#define ENABLE_LEVELING_AFTER_G28

bakaufman, 2021-01-03:
there is a good discussion about this illustrating the confusion around it here: https://github.com/MarlinFirmware/Marlin/issues/20478#issuecomment-753449587

zonalimitatore, 2021-01-03:
Imho I think the first it's for normal leveling, the second it's for UBL that load the mesh

thinkyhead, 2021-02-02:
We're working on automating the generation of configuration documentation. Watch this space.

closed 2021-02-02 by thinkyhead