MarlinFirmware/MarlinDocumentation #381: M122 -add explanation for continuous output

Teddyz, 2021-03-05:
Would be great to have some explanation of what the output means.
Here I do a `G28` homing:

```
X:13/12/-|GTI|	Y:19/10/-|GTI|	Z:12/12/-|GTI|	Z2:13/12/-|GTI|	
X:13/12/-|GTI|	Y:19/10/-|GTI|	Z:12/12/-|GTI|	Z2:13/12/-|GTI|	
X:13/12/-|GTI|	Y:19/10/-|GTI|	Z:50/25/-|G|	Z2:51/25/-|G|	
X:13/12/-|GTI|	Y:19/10/-|GTI|	Z:23/25/-|GTI|	Z2:26/25/-|GTI|	
X:28/25/-|GT|	Y:19/10/-|GTI|	Z:30/25/-|GTI|	Z2:25/25/-|GTI|	
echo:busy: processing
X:28/25/-|GT|	Y:19/10/-|GTI|	Z:26/25/-|GTI|	Z2:28/25/-|GTI|	
X:26/25/-|GTI|	Y:38/20/-|GT|	Z:24/25/-|GTI|	Z2:26/25/-|GTI|	
X:26/25/-|GTI|	Y:39/20/-|GT|	Z:29/25/-|GTI|	Z2:26/25/-|GTI|	
X:25/25/-|GTI|	Y:38/20/-|GTI|	Z:52/25/-|G|	Z2:50/25/-|G|	
echo:busy: processing
X:26/25/-|GTI|	Y:35/20/-|GTI|	Z:24/25/-|GTI|	Z2:31/25/-|GT|	
X:25/25/-|GTI|	Y:37/20/-|GTI|	Z:26/25/-|G|	Z2:28/25/-|G|	
X:25/25/-|GTI|	Y:40/20/-|GTI|	Z:26/25/-|G|	Z2:28/25/-|G|	
X:22/22/-|GTI|	Y:36/20/-|GTI|	Z:26/25/-|G|	Z2:28/25/-|G|	
X:-15.00 Y:-6.00 Z:4.00 E:0.00 Count X:-1200 Y:-600 Z:1600
ok
X:20/20/-|GTI|	Y:40/20/-|GTI|	Z:26/25/-|GTI|	Z2:25/25/-|GTI|	
X:18/18/-|GTI|	Y:35/19/-|GTI|	Z:25/25/-|GTI|	Z2:26/25/-|GTI|	
X:16/16/-|GTI|	Y:32/16/-|GTI|	Z:24/25/-|GTI|	Z2:26/25/-|GTI|	
X:13/13/-|GTI|	Y:28/14/-|GTI|	Z:26/25/-|GTI|	Z2:26/25/-|GTI|	
X:12/12/-|GTI|	Y:22/12/-|GTI|	Z:27/25/-|GTI|	Z2:26/25/-|GTI|	
```

thinkyhead, 2021-04-15:
I haven't got a clue at this time and I'd have to spend some hours to research and figure out the meaning of all the flags shows (and others not shown). Truly, anyone could do this by poring through the Trinamic code and taking lots of notes as I would, so it's possible this will get written up before I have a chance to get around to it.

shitcreek, 2021-04-25:
@Teddyz, could you clarify? Thanks

thinkyhead, 2021-04-28:
`tmc_util.cpp` => `monitor_tmc_drivers` => `monitor_tmc_driver` => `report_polled_driver_data` =>
```cpp
  template<typename TMC>
  void report_polled_driver_data(TMC &st, const TMC_driver_data &data) {
    const uint32_t pwm_scale = get_pwm_scale(st);
    st.printLabel();
    SERIAL_CHAR(':'); SERIAL_PRINT(pwm_scale, DEC);
    #if ENABLED(TMC_DEBUG)
      #if HAS_TMCX1X0 || HAS_TMC220x
        SERIAL_CHAR('/'); SERIAL_PRINT(data.cs_actual, DEC);
      #endif
      #if HAS_STALLGUARD
        SERIAL_CHAR('/');
        if (data.sg_result_reasonable)
          SERIAL_ECHO(data.sg_result);
        else
          SERIAL_CHAR('-');
      #endif
    #endif
    SERIAL_CHAR('|');
    if (st.error_count)       SERIAL_CHAR('E'); // Error
    if (data.is_ot)           SERIAL_CHAR('O'); // Over-temperature
    if (data.is_otpw)         SERIAL_CHAR('W'); // over-temperature pre-Warning
    #if ENABLED(TMC_DEBUG)
      if (data.is_stall)      SERIAL_CHAR('G'); // stallGuard
      if (data.is_stealth)    SERIAL_CHAR('T'); // stealthChop
      if (data.is_standstill) SERIAL_CHAR('I'); // standstIll
    #endif
    if (st.flag_otpw)         SERIAL_CHAR('F'); // otpw Flag
    SERIAL_CHAR('|');
    if (st.otpw_count > 0) SERIAL_PRINT(st.otpw_count, DEC);
    SERIAL_CHAR('\t');
  }
```