MarlinFirmware/MarlinDocumentation #392: M109 and M190 documentation is ambiguous

E1k3, 2021-04-06:
When M109 or M190 are used without specifying a new temperature - for example to wait for bed and nozzle heating at the same time, it isn't specified if Marlin only waits for heating or for cooling as well.

Scenario:
1. I want to set the bed to 60°C and the nozzle to 180°C and heat them up / cool them down to this temperature in parallel
2. M104 S180
3. M140 S60
4. M109
5. M190

Will this result in the nozzle and bed heating / cooling in parallel until both have hit their target temperatures of 180°C and 60°C respectively?

Note that `M109 R180` `M190 R60` is not what is needed here, because in this case, the nozzle has to reach its target before the bed starts to change temperature.

E1k3, 2021-04-09:
This is unrelated to the documentation, but since you probably know - is there any way to wait for nozzle and bed heating/cooling in parallel?

FanDjango, 2021-04-09:
@E1k3 Set Bed temperature with nowait, then set temp with wait for nozzle which is nearly always the faster one and then set same temp with wait for the bed and wait for the bed.