MarlinFirmware/MarlinDocumentation #439: M48 should be called Probe Precision (or Repeatability) Test

cygnyet, 2022-02-24:
Note sure if this is the correct place for this but it does say "Marlin Documentation" so here goes:

Can we change the documentation for M48 so that it is called the " Probe Repeatability Test" (matches the define "Z_MIN_PROBE_REPEATABILITY_TEST").  Also, we should change the body of the documentation so that instead of saying "This command tests the probe for accuracy..." instead it should say "This command tests the probe for repeatability (precision)..."  I particularly think the documentation should state both "repeatability" as the common term and also "precision" which is the more "engineering" term.

It seems many folks think this command can demonstrate that their probe measurements are accurate (give a correct measurement value) when actually all the test can prove is that the probe gives the same (but perhaps wrong) measurement value upon back-to-back repeated measurements,  I have reason to believe (can share data if anyone is interested) that some probes are in fact quite precise but woefully inaccurate.  It is that inaccuracy that can lead to a poor mesh point value measurements and therefore poor auto bilinear mesh leveling.  Poor auto bilinear mesh leveling can be easily observed using a commonly practiced bed leveling test that is the printing of an array of single-layer squares (same pattern as the mesh points) and checking to see that all the squares have acceptable "squish".  Happy to discuss this and the general topic of manual and auto bilinear mesh bed leveling to any like-minded folks.

descipher, 2022-02-24:
I think it would be a probe precision test. It is reporting a deviation between each probing. But yes some may think its an accuracy test, in some aspects you could say it reveals the probes accuracy limit.

cygnyet, 2022-02-24:
My that was a quick response!

In discussions with others, I have found that folks seem to immediately understand the term "repeatability" but their eyes glaze over with "precision".  So, I think it is important to lead with repeatability but include precision for the engineer-types.

Regarding accuracy, I have been spending a lot of time lately explaining the difference between accuracy and repeatability (precision) but I do not understand what you mean by accuracy limit.  Do you mean the resolution of the probe (which is different from both accuracy and precision)?

descipher, 2022-02-24:
Nope, it is as described.
See https://wtmaths.com/applying_limits_accuracy.html

cygnyet, 2022-02-24:
That link is not about precision but is about the uncertainty in a measured value and the limits that are placed on accuracy as a result.   I would therefore still maintain that M48 is only about probe precision and says nothing about probe accuracy or the limits thereof.

descipher, 2022-02-24:
"However it is stated, **the range is the Limit of Accuracy** for that measurement."
"deviation is range"

thisiskeithb, 2022-02-25:
#440 has been merged.

closed 2022-02-25 by thisiskeithb