MarlinFirmware/MarlinDocumentation #447: I'd like to translate the Configutation Guide located on Marlinfw.org

McSloverd, 2022-04-18:
Hi, I would like to translate the Configuration Guide located at [[Here](https://marlinfw.org/docs/configuration/configuration.html)] into Simplified Chinese. Is there anything I should know about?

Valiance, 2022-04-18:
Not really.  Marlin is a big piece of software that interfaces to
hardware.  Murphy's law applies here: 'Whatever can go wrong, will go
wrong.  At the worst possible moment."

McSloverd, 2022-04-19:
> Not really.  Marlin is a big piece of software that interfaces to
> hardware.  Murphy's law applies here: 'Whatever can go wrong, will go
> wrong.  At the worst possible moment."
> 
> 
> > Hi, I would like to translate the Configuration Guide located at [Here
> > <https://marlinfw.org/docs/configuration/configuration.html>] into
> > Simplified Chinese. Is there anything I should know about?
> >
> > <https://github.com/MarlinFirmware/MarlinDocumentation/issues/447>, or
> > unsubscribe

Haha! Thanks for the reply. It exactly illustrates my daily situation.

What I actually meant was "DonI have the permission to translate? Or should I include any specific lines to show the original copyright or license?"

I just want to make it Chinese so that maybe it is easier for people who have language barrier.

I have already finished the first part - Configuration.h but not yet put out to public. I can also give it to Marlin Official if they want it. The first part is around 50k Chinese Characters.

Valiance, 2022-04-19:
I believe what you are doing is a great thing!   I would imagine you will
need to get in touch with the staff that run Marlin directly.

McSloverd, 2022-04-20:
> I believe what you are doing is a great thing! I would imagine you will need to get in touch with the staff that run Marlin directly.
> […](#)

Thanks for the reply! I believe so too and I have tried to find any contact method to reach the staff and I couldn't even find an email address. I thought if I open an issue here maybe someone from the staff could see it.

I have started the translation of Configuration_adv.h today.

Valiance, 2022-04-20:
Here is a link to Marlin Discord channel, perhaps you can get their
attention here.
https://discord.gg/b2BQWwyt

McSloverd, 2022-04-21:
> Here is a link to Marlin Discord channel, perhaps you can get their
> attention here.
> https://discord.gg/b2BQWwyt
> 
> 
> 
> 
> > I believe what you are doing is a great thing! I would imagine you will
> > need to get in touch with the staff that run Marlin directly.
> > … <#m_9148447586547413476_>
> > On Mon, Apr 18, 2022 at 9:55 PM Simeng Fu *@*.*> wrote: Not really.
> > Marlin is a big piece of software that interfaces to hardware. Murphy's law
> > applies here: 'Whatever can go wrong, will go wrong. At the worst possible
> > moment." On Mon, Apr 18, 2022 at 2:43 AM Simeng Fu @.*> wrote: Hi, I
> > would like to translate the Configuration Guide located at [Here
> > https://marlinfw.org/docs/configuration/configuration.html] into
> > Simplified Chinese. Is there anything I should know about? — Reply to this
> > email directly, view it on GitHub #447
> > <https://github.com/MarlinFirmware/MarlinDocumentation/issues/447> <#447
> > <https://github.com/MarlinFirmware/MarlinDocumentation/issues/447>>, or
> > unsubscribe
> > situation. What I actually meant was "DonI have the permission to
> > translate? Or should I include any specific lines to show the original
> > copyright or license?" I just want to make it Chinese so that maybe it is
> > easier for people who have language barrier. I have already finished the
> > first part - Configuration.h but not yet put out to public. I can also give
> > it to Marlin Official if they want it. The first part is around 50k Chinese
> > (comment)
> > <https://github.com/MarlinFirmware/MarlinDocumentation/issues/447#issuecomment-1101916033>>,
> >
> > Thanks for the reply! I believe so too and I have tried to find any
> > contact method to reach the staff and I couldn't even find an email
> > address. I thought if I open an issue here maybe someone from the staff
> > could see it.
> >
> > I have started the translation of Configuration_adv.h today.

Thanks for pointing me to the direction. I have joined the discord, however there are quite a lot of channels and I couldn't tell which one is a good one for me to ask such question. Do you have any suggestion? Thanks!

ellensp, 2022-04-21:
All posts in Marlin github are already re posted in discord channels.

I have some concerns,
1) who will maintain it once your board of it?  (this is inevitable) 
2) there is nothing in place to support language choices (though a independent page is an option)
3) I'm told google translate does a fairly good job (I use it for French and German pages quite often) this will always be up to date with main page. (vs someone having to update each version of the guide every time the English guide is updated)

@thinkyhead Can we get your views on this?

McSloverd, 2022-04-21:
> All posts in Marlin github are already re posted in discord channels.
>
>
>
> I have some concerns,
>
> 1) who will maintain it once your board of it?  (this is inevitable)
>
> 2) there is nothing in place to [...]

Thanks for the reply! 

1. I can maintain it as long as I am asked to. Or my company which is basically working with marlin everyday. I can assign this task to a small group of staff. 

2. I was originally planning to "publish" it on Zhihu.com - a Chinese alternative of Reddit / quora. Quite a place for professionals. But if we are discussing long-term professional maintaining, I can get a domain, like "marlinfw.cn" and put it there. And yes, marlin officials can have the ownership. I will pay for everything, as a tiny contribution to the fantastic community.

3. Google (and essentially other big offshore online service) is almost all banded by the "Great Wall", I can put it on Google, but a lot of Chinese will never see it. This basically defeats the original purpose of what I am trying to do. If I put it domestically (in China), not just Chinese can get access to it but also everyone in the world could as well. Of course, to make this happen, not only the domain name and the storage of the content has to be located in China, but also permission and licenses has to be granted. I can arrange these license and just like I have mentioned, the cost is on me.

Valiance, 2022-04-21:
I would send you to post in the General tab of the Marlin Discord channel.
Post what you want to accomplish there, and the Devs will get wind of it
soon enough.

McSloverd, 2022-04-22:
I would show you guys how it look like but I have not yet publish it. This is a screen shot of it.

I tried to keep the exact structure of the original document.

[image]

McSloverd, 2022-04-22:
> I would send you to post in the General tab of the Marlin Discord channel.
> Post what you want to accomplish there, and the Devs will get wind of it
> soon enough.
> 
> 
> > All posts in Marlin github are already re posted in discord channels.
> >
> > I have some concerns,
> >
> >    1.
> >
> >    who will maintain it once your board of it? (this is inevitable)
> >    2.
> >
> >    there is nothing in place to support language choices (though a
> >    independent page is an option)
> >    3.
> >
> >    I'm told google translate does a fairly good job (I use it for French
> >    and German pages quite often) this will always be up to date with main
> >    page. (vs someone having to update each version of the guide every time the
> >    English guide is updated)
> >
> > @thinkyhead <https://github.com/thinkyhead> Can we get your views on this?
> >
> > Thanks for the reply!
> >
> >    1.
> >
> >    I can maintain it as long as I am asked to. Or my company which is
> >    basically working with marlin everyday. I can assign this task to a small
> >    group of staff.
> >    2.
> >
> >    I was originally planning to "publish" it on Zhihu.com - a Chinese
> >    alternative of Reddit / quora. Quite a place for professionals. But if we
> >    are discussing long-term professional maintaining, I can get a domain, like
> >    "marlinfw.cn" and put it there. And yes, marlin officials can have the
> >    ownership. I will pay for everything, as a tiny contribution to the
> >    fantastic community.
> >    3.
> >
> >    Google (and essentially other big offshore online service) is almost
> >    all banded by the "Great Wall", I can put it on Google, but a lot of
> >    Chinese will never see it. This basically defeats the original purpose of
> >    what I am trying to do. If I put it domestically (in China), not just
> >    Chinese can get access to it but also everyone in the world could as well.
> >    Of course, to make this happen, not only the domain name and the storage of
> >    the content has to be located in China, but also permission and licenses
> >    has to be granted. I can arrange these license and just like I have
> >    mentioned, the cost is on me.

Thanks! I will give a look at the channel. Honestly, I have never used discord before. Guess I am getting old😹

thinkyhead, 2022-04-22:
My hope is to produce a nice script in Python or Perl that will convert the files `Configuration.h` and `Configuration_adv.h` (and maybe elements of `SanityCheck.h`) into Jekyll, with a separate page for each top-level option, and sub-options documented under those options.

So the first step in a Chinese translation would be to translate all the documentation in those files and post a Chinese set of configuration files. Those could be maintained as a bonus within the `MarlinFirmware/Configurations` repository.

Then, whenever I manage to write that Config-to-Jekyll script (and from that time onward) I can generate both English and Chinese pages at the same time.

McSloverd, 2022-04-22:
> > Can we get your views on this?
> 
> 
> 
> My hope is to produce a nice script in Python or Perl that will convert the files `Configuration.h` and `Configuration_adv.h` (and maybe elements of `SanityCheck.h`) into Jekyll, with a separate page for each top-level option, and sub-options documented under those options.
> 
> 
> 
> So the first step in a Chinese translation would be to translate all the documentation in those files and post a Chinese set of configuration files. Those could be maintained as a bonus within the `MarlinFirmware/Configurations` repository.
> 
> 
> 
> Then, whenever I manage to write that Config-to-Jekyll script (and from that time onward) I can generate both English and Chinese pages at the same time.

Thanks for your reply!

That would be awesome! I presume that this would be taking some time, right? I propose that I can start the translation of that web page and put it up online. And when you start this python project I can always participate to contribute the translation work, if you need of course. And since China is quite a market for 3D printing, I propose that I can translate the whole marlinfw.org into Chinese and put it to marlinfw.cn, for example. Of course all these work and cost is for free as my contribution to the community, and the ownership of all related work goes to marlinfw.org. I have benefited from marlin and I just want to give something back, to both Marlin and Chinese 3D Printing.

Thanks!