MarlinFirmware/MarlinDocumentation #466: Character LCDs and Graphical LCDs out out of date.

StanislavJochman, 2022-09-08:
Hi some LCDs are out of date and links no longer works. For example anet displays.

Best regards
Stanislav Jochman

ellensp, 2022-09-09:
Suspect they are talking about https://marlinfw.org/docs/configuration/configuration.html#lcd-controller

StanislavJochman, 2022-09-09:
> Suspect they are talking about https://marlinfw.org/docs/configuration/configuration.html#lcd-controller

Yeah you are right

thinkyhead, 2022-11-04:
This falls under "general update of the Configuration pages" which are admittedly a bit of a hassle to maintain. We hope to build scripts to automate the process, at least to create the initial pages for existing options. That work is in progress as part of our schema extraction scripts, where the next step is to use the configuration data to produce hierarchical YAML that will be turned into pages and indexes within the Jekyll site. See the current work on scripts within MarlinFirmware's "buildroot" folder.

closed 2022-11-04 by thinkyhead