MarlinFirmware/MarlinDocumentation #493: [FR] Linear Advance Pattern - line through the numbers on the side

Anarasha, 2023-04-02:
### Is your feature request related to a problem? Please describe.

This is not technically a firmware request as it is a request for the linear advance calibration pattern generator(https://marlinfw.org/tools/lin_advance/k-factor.html) 

**The mandatory praise:**
I am a huge fan of this tool, whoever designed it needs to be immortalized in a statue or a song. 

**The "problem":**
Removing all the numbers from my bed is a massive chore, and I believe that a small feature added could make the removal much easier. 

### Are you looking for hardware support?

Nothing hardware specific

### Describe the feature you want

**My feature request/proposed solution:**
I would like an option to add a line to both sides of the numbering, like the anchor frames for the lines themselves, to connect all the numbers. 

It would also be worth considering making said option also do a line through the bottom of the numbers to connect all the numbers in one piece.  I realize this would make the legibility of the numers a bit less clear, but the convenience of removal afterwards would more than make up for that 

### Additional context

Removing the numbers when testing 0-0.5 in two prints with 0.01 steps gets real tedious and they get _everywhere_. I'm pretty sure I just found one in my ear from 6 moths ago. I definitely just found one depressed into my magnetic mat.

thinkyhead, 2023-08-07:
Transferred to #517

closed 2023-08-07 by thinkyhead

thisiskeithb, 2023-08-07:
The issue was transferred here because the work for this PR needs to happen in this repo (MarlinDocumentation) as this repo drives the code behind marlinfw.org.