MarlinFirmware/MarlinDocumentation #495: Input Shaping dosent work in PrusaSlicer

MrBrain295, 2023-04-17:
Putting `M593 F{(layer_num < 2 ? 0 : 15 + 45.0 * (layer_num - 2) / 297)} ; Hz Input Shaping Test` as the layer change GCode in PrusaSlicer 2.5.1 gives this error: Post-processing script M593 F{(layer < 2 ? 0 : 15 + 45.0 * (layer - 2) / 297)} on file /tmp/.22125.gcode.pp failed.
Error code: 2
Output:
/bin/bash: -c: line 1: syntax error near unexpected token `('
/bin/bash: -c: line 1: `M593 F{(layer < 2 ? 0 : 15 + 45.0 * (layer - 2) / 297)} '/tmp/.22125.gcode.pp''

The-EG, 2023-04-17:
Where exactly did you put the given code? Your error would indicate you put it in the post processing script instead of after layer change gcode.

MrBrain295, 2023-04-17:
Yeah I noticed that but I put it in the layer change gcode. It also showed
up in the post processing scripts section although I didn’t put it there. I
thought it was a part of prusaslicer but it works now after removing it.
Thanks.

closed 2023-04-17 by MrBrain295