MarlinFirmware/MarlinDocumentation #532: Issue with K-factor Calibration Pattern

kendrickxy, 2023-12-05:
When I enter a value that causes an error, the software won't let me generate the G-code, even after correction.
i.g:

[image]

[image]

I get this error. So I change the K-Factor Stepping to 0.002, but I'm still unable to generate the G-code.

The only solution I've found is to reload the page and enter the data once again.

kendrickxy, 2023-12-05:
Well I guess changing the html code is also a work around 😄

thisiskeithb, 2023-12-05:
The form doesn't re-validate settings until you tab/click into another textbox.

kendrickxy, 2023-12-09:
I've tried, Textbox, Dropbox, checkbox. Unfortunately doesn't fix the Generate button

thisiskeithb, 2023-12-09:
Which OS & browser are you using?

thinkyhead, 2023-12-12:
Unable to reproduce…
[image]

Floating point maths can sometimes be imprecise. So it's possible the test for even divisibility fails on some devices with wonky math libraries. Try different browsers and operating systems to see if they do the same thing. Meanwhile, we can try adjusting this test to see if adding `parseInt` or other tricks help.

kendrickxy, 2023-12-12:
I am using Edge, but in the video I tried Chrome.

https://github.com/MarlinFirmware/MarlinDocumentation/assets/16179839/5db2609e-05cd-405a-881a-465f2d52b322

Am I doing it wrong?

thinkyhead, 2024-01-05:
The buttons should be fixed now.

closed 2024-01-05 by thinkyhead