MarlinFirmware/MarlinDocumentation #563: A question regarding the M600 parameters E, U, and L

GregValiant, 2024-06-09:
### Did you test the latest `bugfix-2.1.x` code?

N\A

### Question

I want to configure M600 on the fly in a UM Cura post-processor.  I am looking for a clarification regarding whether the E, U and L parameters should be entered as positive or negative values.  The information I've found appears to be contradictory.

The M600 command parameters E, U, L are listed on the [Gcodes page](https://marlinfw.org/meta/gcode/) as requiring negative values.  It seems odd that the L (load filament) parameter would also require a negative value.
"[E<pos>] | Retract before moving to change position (**negative**, default PAUSE_PARK_RETRACT_LENGTH)"
"[U<pos>] | Amount of retraction for unload (**negative**)"
"[L<pos>] | Load length, longer for bowden (**negative**)"
It appears that configured like this:
M600 B1 **E-6.5 U-450** R205 T0 **L-30** X0 Y0 Z2
there will be an E movement of -6.5mm (retraction), an unload of -450mm, and a reload of -30mm??

These are from M600.cpp:
 *  E[distance] - Retract the filament this far
 *  U[distance] - Retract distance for removal (manual reload)
 *  L[distance] - Extrude distance for insertion (manual reload)
 Those make it seem like they should all 3 should be positive values.

Then later in M600.cpp these appear to be switching the +/- signs:
 * -ABS(parser.axisunitsval('E', E_AXIS, PAUSE_PARK_RETRACT_LENGTH))
 * -ABS(parser.axisunitsval('U', E_AXIS, fc_settings[active_extruder].unload_length)) : 0.5f;
 *  ABS(parser.axisunitsval('L', E_AXIS, fc_settings[active_extruder].load_length))
 That appears that they are a mix and get changed later...or something.

If someone could take a moment to clarify the proper input for "Retract, Unload, and Load" I'd be much obliged.

GregValiant

### Bug Timeline

N\A

### Expected behavior

N/A

### Actual behavior

N/A

### Steps to Reproduce

N\A

### Version of Marlin Firmware

N\A

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

N\A

thisiskeithb, 2024-06-09:
I agree that documentation can be cleared up a bit, but you can use positive or negative values.

Marlin will use the absolute value (`ABS()`), but unloads/retracts need to be negative and loads/extrudes need to be positive, so that is done automatically with `-ABS()` or `ABS()`.

The same goes for [`M603 - Configure Filament Change`](https://marlinfw.org/docs/gcode/M603.html). If you enter  `M603 U120 L125` or `M603 U-120 L-125`, the absolute values are used/reported, but with `fc_settings[target_extruder].unload_length` & `fc_settings[target_extruder].load_length`:

#### Unload length
```c
  if (parser.seenval('U')) {
    fc_settings[target_extruder].unload_length = ABS(parser.value_axis_units(E_AXIS));
```

#### Load length
```c
  if (parser.seenval('L')) {
    fc_settings[target_extruder].load_length = ABS(parser.value_axis_units(E_AXIS));
```

GregValiant, 2024-06-09:
Thanks for that.  The question is in regards to the Cura "FilamentChange" post processor.  I'll clear it up the the tooltip.

closed 2024-08-19 by thinkyhead